facebookresearch/habitat-sim #1080: Question about using gibson tiny dataset for habitat-sim

uesuckbabo, 2021-02-09:
## ❓ Questions and Help
Hello, I want to use gibson-tiny dataset for habitat-sim as shown in the paper "object goal navigation using goal-oriented semantic exploration" because I need semantic annotations for my research and gibson dataset currently only supports it for tiny and medium datasets. Since the gibson tiny dataset is for the gibson environment, the dataset needs to be converted for use in habitat-sim. I have downloaded the gibson tiny dataset and followed the steps in #54 and ran the ./build/viewer  --path/to/scene_dataset.glb only to see a black screen. (Converting from obj to glb is successful and the glb can be loaded in /build/viewer but the view is completely black) I found that unlike the case in #54,   there is no mtl file in the gibson tiny dataset but I can't seem to find a solution to how to solve this problem. 

I would like to know how I can use gibson-tiny dataset for habitat-sim

erikwijmans, 2021-02-09:
Please run the script as specified here: https://github.com/facebookresearch/habitat-sim#datasets.  After that, you just point habitat at the normal glb used for a given scene.  

The reason you are seeing black is likely because you converted the semantic objs to glbs and then are trying to render them, they have no RGB values and thus black.

uesuckbabo, 2021-02-09:
From what I have understood, the link https://github.com/facebookresearch/habitat-sim#datasets is used to create the semantic.ply file which I have already done.
I have converted mesh.objs file to glb in this case which turned out black because it has no rgb values. So I was wondering how to solve this problem. Can you give a longer explanation about this part "After that, you just point habitat at the normal glb used for a given scene."

erikwijmans, 2021-02-09:
There are the normal GLBs that get downloaded as part of the Gibson dataset for habitat.  That download includes the glbs needed to do RGB rendering for Gibson tiny.

closed 2021-08-16 by dhruvbatra

labixiaochao, 2025-09-02:
> There are the normal GLBs that get downloaded as part of the Gibson dataset for habitat. That download includes the glbs needed to do RGB rendering for Gibson tiny.

Have you been successful in downloading and creating Gibson? I also encountered some problems, can we add a contact information to communicate?

labixiaochao, 2025-09-02:
> ## ❓ Questions and Help
>
> Hello, I want to use gibson-tiny dataset for habitat-sim as shown in the paper "object goal navigation using goal-oriented semantic exploration" because I need semantic [...]

Have you been successful in downloading and creating Gibson? I also encountered some problems, can we add a contact information to communicate?