facebookresearch/habitat-sim #1084: Segmentation fault (core dumped) on executing python examples/example.py

LaughBuddha, 2021-02-10:
## ❓ Questions and Help
On executing the command `python examples/example.py --scene ~/habitat-api/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

I'm getting the below message - 
python examples/example.py --scene ~/habitat-api/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
```
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0210 02:01:48.044404 208131 WindowlessContext.cpp:98] [EGL] Detected 5 EGL devices
Segmentation fault (core dumped)
```

jturner65, 2021-02-10:
Hello
What version of habitat-sim and habitat-lab are you using?  It seems like you might be using an old version of habitat-lab (habitat-api was changed to habitat-lab last year) and this could be the issue.

erikwijmans, 2021-02-10:
What Nvidia driver version are you on?  There is a known issue with `450.80.02` that causes this https://github.com/facebookresearch/habitat-sim/pull/1006

LaughBuddha, 2021-02-17:
> Hello
> What version of habitat-sim and habitat-lab are you using? It seems like you might be using an old version of habitat-lab (habitat-api was changed to habitat-lab last year) and this could be the issue.

Yes, I am using the below checkout for habitat-sim and habitat-api for a specific requirement
habitat-sim: 9575dcd45fe6f55d2a44043833af08972a7895a9
habitat-api: b5f2b00a25627ecb52b43b13ea96b05998d9a121

LaughBuddha, 2021-02-17:
> What Nvidia driver version are you on? There is a known issue with `450.80.02` that causes this #1006

I am using the below NVIDIA driver
[image]

erikwijmans, 2021-02-17:
There's a good chance that that driver still has the bug.  I don't have a machine with that driver version, but it is very likely given where you are segfaulting.

Either update habitat-sim or backprot the workaround.  It's just two lines
```cpp
const char* const eglExtensions = eglQueryDeviceStringEXT(devices[selectedDevice], EGL_EXTENSIONS);
if(eglGetError() == EGL_BAD_DEVICE_EXT) continue
```
That needs to go before `eglQueryDeviceAttribEXT(devices[selectedDevice], EGL_CUDA_DEVICE_NV, &cudaDeviceNumber)` in `WindowlessContext.cpp`.  Variable names may be slightly different in the older code, but the function names are the same.

LaughBuddha, 2021-02-17:
Hi @erikwijmans 

I made the above code change as suggested in `WindowlessContext.cpp`

I am getting the below error now -
[image]

erikwijmans, 2021-02-17:
Please take a look a https://github.com/facebookresearch/habitat-sim/issues/288

LaughBuddha, 2021-02-17:
Hi Erik,

Below is the output I get after running `ldconfig -N -v | grep libEGL`
[image]

erikwijmans, 2021-02-18:
Please pull in this to get rid of gflags: https://github.com/facebookresearch/habitat-sim/pull/247

LaughBuddha, 2021-02-18:
Hi Erik,

Thanks for your reply and patience with this issue.

But I'm still getting the same error after making the changes mentioned in #247 
```
(ActiveNeuralSLAM) sidgoel@node-1080ti-0:~/habitat-api$ python examples/example.py
2021-02-17 20:34:08,024 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/mnt/beegfs/home/sidgoel/habitat-api/habitat/core/env.py", line 93, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/mnt/beegfs/home/sidgoel/habitat-api/habitat/sims/registration.py", line 19, in make_sim
    return _sim(**kwargs)
  File "/mnt/beegfs/home/sidgoel/habitat-api/habitat/sims/habitat_simulator/__init__.py", line 24, in __init__
    raise habitat_sim_import_error
  File "/mnt/beegfs/home/sidgoel/habitat-api/habitat/sims/habitat_simulator/__init__.py", line 7, in _try_register_habitat_sim
    import habitat_sim
  File "/home/sidgoel/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/sidgoel/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/sidgoel/habitat-sim/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/sidgoel/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/sidgoel/habitat-sim/habitat_sim/agent/agent.py", line 14, in <module>
    import habitat_sim.bindings as hsim
  File "/home/sidgoel/habitat-sim/habitat_sim/bindings/__init__.py", line 24, in <module>
    from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
ImportError: libgflags.so.2.2: cannot open shared object file: No such file or directory
```

erikwijmans, 2021-02-18:
Try a clean rebuild, i.e. `rm -rf build`

LaughBuddha, 2021-02-18:
Resolved.

closed 2021-02-18 by LaughBuddha