facebookresearch/habitat-sim #1085: Is Gibson Semantic Label supposed to be sparse?

ReNginx, 2021-02-10:
## ❓ Questions and Help

Hi, 

I would like to use habitat to render semantics of Gibson. I followed the instructions in #263 to convert instance id to category id.

The semantic label I got from Gibson dataset is extremely sparse, which seems not consistent with what Gibson's front page shows.

Below is my result. 
[image]

and below is the label demo from Gibson's github [page ](https://github.com/StanfordVL/GibsonEnv/blob/master/gibson/utils/semantics.md)

[image]

The same code works well on matterport. Like the image below.

[image]

I'm wondering what might be the problems?

Thank you for the help!!!

ReNginx, 2021-02-10:
I ran across #910, which also happened on me.

after loading scene `Adairsville` , and running `print(sim.semantic_scene.objects)`, I got the following result.

```
[None, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6d30>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6d70>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6c30>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6c70>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6ab0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6af0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6870>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a69f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a69b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6970>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a6730>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4a67b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2b1a1b30>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef88f7a270>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4af970>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4aff30>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef2a4affb0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6030>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6070>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc60b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc60f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6130>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6170>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc61b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc61f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6230>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6270>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc62b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc62f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6330>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6370>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc63b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc63f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6430>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6470>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc64b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc64f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6530>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6570>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc65b0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc65f0>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6630>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc6670>, <habitat_sim._ext.habitat_sim_bindings.SemanticObject object at 0x7fef0ffc66b0>]
```

Is the `None` object a bug that caused the loss of semantic labels in Gibson?

erikwijmans, 2021-02-10:
Yes, the semantic labels for Gibson are very sparse.  The link you showed is their rendering on Matterport3D semantic labels.  This is their rendering of the Gibson labels: https://3dscenegraph.stanford.edu

None objects are not a bug, they are a necessary side effect of storing objects as a list but wanting the object's position in that list to be its instance ID.

ReNginx, 2021-02-14:
Thank you very much for the quick response!

closed 2021-02-14 by ReNginx