facebookresearch/habitat-sim #1092: how to get frames?

yunsujeon, 2021-02-15:
## ❓ Questions and Help
hi, 
I wanna detect yolov4 on habitat-sim. The first step is get habitat-sim's frame. but i don't have any ideas..

i installed habitat-sim (conda) , ./build.sh and run this line
`habitat-viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

how to I get the real time frame in Interactive viewer? to use yolo

dhruvbatra, 2021-02-15:
Please go through the Habitat Sim basics tutorials:

https://aihabitat.org/tutorial/2020/

closed 2021-02-15 by dhruvbatra

yunsujeon, 2021-02-15:
@dhruvbatra 
Umm.. that tutorial is run by python code and I understood that means .. User must write a script. (go left, go to bathroom...else)
but i want run "habitat-viewer"and move by "WASD" control. no robot navigation.
In WASD moving, can i get frames?

erikwijmans, 2021-02-15:
Pressing the `i` key will save a screenshot.  The viewer isn't designed as a data collection tool and we don't have plans to make it one, so you may end up just fighting with it.  As Dhruv suggested, going through the python API will likely be easier.  Another thing to look at is https://aihabitat.org/docs/habitat-sim/image-extractor.html