facebookresearch/habitat-sim #1101: System Requirements?

yc-chn, 2021-02-20:
## ❓ Questions and Help
os version? cpu cores? ram? gpu? vram? storage?
Minimum Spec, Good, Ideal?
not found in docs, not found in readme, not found in aihabitat.org homepage

erikwijmans, 2021-02-20:
We don't give these as it's so dependent on what you want to do that it is impossible to give meaningful recommendations.  In general, any machine that can do a reasonable job at DL training is fine.

For OS, we support linux flavor and MacOS.  Folks have gotten it compiling on windows (as magnum supports that), but we don't test it.

closed 2021-02-20 by erikwijmans

dman82499, 2021-02-25:
will windows ever be supported in the future?

Skylion007, 2021-02-25:
It is not a priority at the moment.

The graphics library should support Windows out of the box, so it's probably just a matter of tweaking the CMake files/build process. Feel free to submit a PR.