facebookresearch/habitat-sim #1109: Habitat Tutorials on Colab are failing.

yashkant, 2021-02-26:
## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. Run any notebooks from -- https://aihabitat.org/tutorial/2020/

 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Additional context
I debugged this issue, it seems like the default Python Version for Colab has changed from 3.6 to 3.7. 
The installation [script](https://raw.githubusercontent.com/facebookresearch/habitat-sim/master/examples/colab_utils/colab_install.sh) used in the tutorial has a hardcoded Python version of 3.6, which links the conda installed libraries to `site-packages`. 

## Fix 
As of now, I am able to get everything running by just changing the version of Python in the installation script from 3.6 to 3.7.
[Here](https://transfer.sh/o6kpR/colab_install.sh) is the script that works for me currently. 

Ideally, we should be inserting this version dynamically by checking the Python version running on Colab Notebook.

Skylion007, 2021-02-26:
I feared this might happen... https://twitter.com/GoogleColab/status/1365062244225060865?s=20

Skylion007, 2021-02-27:
Closed by #1110

closed 2021-02-27 by Skylion007