facebookresearch/habitat-sim #1132: Error in running examples.py for habitat-sim version

LaughBuddha, 2021-03-08:
## ❓ Questions and Help
Hi,

After installing the `ae6ba1cdc772f7a5dedd31cbf9a5b77f6de3ff0f` version of habitat-sim, I run the following test script to test the installation.
`python examples/example.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`

 But i'm getting the below error message. 
`ImportError: ImportError: /mnt/beegfs/home/sidgoel/multiOn/lib/python3.6/site-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: file too short`

Also, getting the same error message on testing the installation for habitat-lab. Below is the complete dump.

```
(multiOn) sidgoel@node-2080ti-5:~/habitat-sim$ python examples/example.py --scene ~/habitat-lab/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
Traceback (most recent call last):
  File "examples/example.py", line 10, in <module>
    import demo_runner as dr
  File "/mnt/beegfs/home/sidgoel/habitat-sim/examples/demo_runner.py", line 15, in <module>
    from settings import default_sim_settings, make_cfg
  File "/mnt/beegfs/home/sidgoel/habitat-sim/examples/settings.py", line 5, in <module>
    import habitat_sim
  File "/mnt/beegfs/home/sidgoel/habitat-sim/habitat_sim/__init__.py", line 14, in <module>
    from habitat_sim.nav import *
  File "/mnt/beegfs/home/sidgoel/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: ImportError: /mnt/beegfs/home/sidgoel/multiOn/lib/python3.6/site-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: file too short

At:
  /mnt/beegfs/home/sidgoel/multiOn/lib/python3.6/site-packages/magnum/__init__.py(28): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(678): exec_module
  <frozen importlib._bootstrap>(665): _load_unlocked
  <frozen importlib._bootstrap>(955): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap>(941): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(922): create_module
  <frozen importlib._bootstrap>(571): module_from_spec
  <frozen importlib._bootstrap>(658): _load_unlocked
  <frozen importlib._bootstrap>(955): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  /mnt/beegfs/home/sidgoel/habitat-sim/habitat_sim/nav/__init__.py(1): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(678): exec_module
  <frozen importlib._bootstrap>(665): _load_unlocked
  <frozen importlib._bootstrap>(955): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  /mnt/beegfs/home/sidgoel/habitat-sim/habitat_sim/__init__.py(14): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(678): exec_module
  <frozen importlib._bootstrap>(665): _load_unlocked
  <frozen importlib._bootstrap>(955): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  /mnt/beegfs/home/sidgoel/habitat-sim/examples/settings.py(5): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(678): exec_module
  <frozen importlib._bootstrap>(665): _load_unlocked
  <frozen importlib._bootstrap>(955): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  /mnt/beegfs/home/sidgoel/habitat-sim/examples/demo_runner.py(15): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(678): exec_module
  <frozen importlib._bootstrap>(665): _load_unlocked
  <frozen importlib._bootstrap>(955): _find_and_load_unlocked
  <frozen importlib._bootstrap>(971): _find_and_load
  examples/example.py(10): <module>
```

erikwijmans, 2021-03-08:
> ImportError: ImportError: /mnt/beegfs/home/sidgoel/multiOn/lib/python3.6/site-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: file too short

That's a filesystem error, please contact your system admins and/or reinstall and hope it works that time.

closed 2021-03-08 by erikwijmans

LaughBuddha, 2021-03-08:
Works as expected upon reinstallation.