facebookresearch/habitat-sim #1143: Conda install for specific tag of habitat-sim

Kritika100, 2021-03-16:
Are there pre-build binaries or conda packages for a specific tag e.g. v0.15 of habitat-sim ?

The following command installs the master or latest version of habitat-sim :
conda install habitat-sim -c conda-forge -c aihabitat

Thanks

erikwijmans, 2021-03-23:
Yes, you can specify the version number, i.e. `condo install habitat-sim=0.1.6 -c conda-forge -c aihabitat`.  We do have binaries for 0.1.4 and 0.1.5, but they have slightly strange version numbers and only python 3.6 as we were doing the initial testing then, you can see the version numbers here: https://anaconda.org/aihabitat/habitat-sim/files

Kritika100, 2021-03-23:
Thanks  @erikwijmans.

closed 2021-03-23 by Kritika100