facebookresearch/habitat-sim #1147: Import error in ECCV_2020_Navigation.ipynb

KenaHemnani, 2021-03-21:
In the following colab notebook : 
https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb   

I am encountering following error in the cell **Colab Setup and Imports** ,whereas the code was working fine before 3 days. Please, give me suggestions to solve this:
`RuntimeError                              Traceback (most recent call last)
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
`
`ImportError                               Traceback (most recent call last)
<ipython-input-4-242787fc7f39> in <module>()
     18 from PIL import Image
     19 
---> 20 import habitat_sim
     21 from habitat_sim.utils import common as utils
     22 from habitat_sim.utils import viz_utils as vut

3 frames
/usr/local/lib/python3.7/dist-packages/habitat_sim/__init__.py in <module>()
     11 if not getattr(builtins, "__HSIM_SETUP__", False):
     12     # TODO: kept only for compatibiliy with existing code. Please gradually remove
---> 13     from habitat_sim import (  # noqa: F401
     14         agent,
     15         attributes,

/usr/local/lib/python3.7/dist-packages/habitat_sim/agent/__init__.py in <module>()
      7 # type: ignore
      8 
----> 9 from .agent import *  # noqa: F401,F403
     10 from .controls import *  # noqa: F401,F403
     11 

/usr/local/lib/python3.7/dist-packages/habitat_sim/agent/agent.py in <module>()
     10 import magnum as mn
     11 import numpy as np
---> 12 import quaternion
     13 
     14 import habitat_sim.errors

/usr/local/lib/python3.7/dist-packages/quaternion/__init__.py in <module>()
     23 import numpy as np
     24 
---> 25 from .numpy_quaternion import (
     26     quaternion, _eps, slerp_evaluate, squad_evaluate,
     27     # slerp_vectorized, squad_vectorized, slerp, squad,

ImportError: numpy.core.multiarray failed to import

---------------------------------------------------------------------------
NOTE: If your import is failing due to a missing package, you can
manually install dependencies using either !pip or !apt.

To view examples of installing some common dependencies, click the
"Open Examples" button below.
---------------------------------------------------------------------------`

Skylion007, 2021-03-21:
Yeah, this is an issue with a badly specificied requirements for a new Conda package. I opened an issue: https://github.com/moble/quaternion/issues/161 . If it doesn't get resolved promptly, we will create a workaround. 

Until then updating numpy in Colab (with conda) should clear the issue or downgrading quaternion to an earllier release (the later is probably easier.

moble, 2021-03-21:
As mentioned on that other issue, this has nothing to do with `quaternion`; it's just `quaternion`'s bad luck that it is the first thing that tries to actually *use* `numpy` after `numpy` is imported.  

This is a fairly common effect of bad installs.  In particular, the output shows that the correct (most recent) version of numpy is actually getting installed, but something about python is failing to find the right version of each part of numpy.  In this case, I would guess that it's because that notebook tries to install everything, and then use it immediately inside colab.  I see, for example, that conda complains about `PYTHONPATH` being set, which is always a bad idea.  In my testing, it's enough to just restart the runtime after that first installation step.  Basically, this is just an inadequate way of installing python.

Skylion007, 2021-03-21:
Yeah, there are some tricks needed install conda packages in Colab without forcing the user to restart. I just pinned the conda version to use the right numpy and it should work for now.

KenaHemnani, 2021-03-22:
> As mentioned on that other issue, this has nothing to do with `quaternion`; it's just `quaternion`'s bad luck that it is the first thing that tries to actually _use_ `numpy` after `numpy` is imported.
> 
> This is a fairly common effect of bad installs. In particular, the output shows that the correct (most recent) version of numpy is actually getting installed, but something about python is failing to find the right version of each part of numpy. In this case, I would guess that it's because that notebook tries to install everything, and then use it immediately inside colab. I see, for example, that conda complains about `PYTHONPATH` being set, which is always a bad idea. In my testing, it's enough to just restart the runtime after that first installation step. Basically, this is just an inadequate way of installing packages with python.

Yeah , Thank you for the quick response. Restarting run time after first installation worked for me . Thanks.

KenaHemnani, 2021-03-22:
> Yeah, there are some tricks needed install conda packages in Colab without forcing the user to restart. I just pinned the conda version to use the right numpy and it should work for now.

Thank you for the quick response.
I just tried the new notebook from the website again from https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb 

But, it is giving same error. May be I didn't understood your solution. Can you elaborate more about it.  Thank you.

aclegg3, 2021-03-22:
Confirmed that restarting the runtime after installation is a work-around for this issue.

PR #1148 (2021-03-22): [COLAB] Pin Numpy Version in Colab install
## Motivation and Context

Fixes #1147

## How Has This Been Tested
With Colab

## Types of changes

- [X] Docs change / refactoring / dependency upgrade
- [X] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

## Checklist

- [X] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [X] I have re [... cut]
closed 2021-03-22 by Skylion007