facebookresearch/habitat-sim #1174: AttributeError: module 'habitat_sim' has no attribute 'SensorSubType'

LaughBuddha, 2021-04-16:
Getting the below error log
```
File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/__init__.py", line 9, in <module>
    from habitat.core.benchmark import Benchmark
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/core/benchmark.py", line 18, in <module>
    from habitat.core.env import Env
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/core/env.py", line 20, in <module>
    from habitat.datasets import make_dataset
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/__init__.py", line 7, in <module>
    from habitat.datasets.registration import make_dataset  # noqa: F401 .
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/registration.py", line 23, in <module>
    _try_register_objectnavdatasetv1()
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/object_nav/__init__.py", line 20, in _try_register_objectnavdatasetv1
    from habitat.datasets.object_nav.object_nav_dataset import (  # noqa: F401
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/object_nav/object_nav_dataset.py", line 15, in <module>
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/__init__.py", line 7, in <module>
    from habitat.tasks.registration import make_task  # noqa: F401
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/registration.py", line 24, in <module>
    _try_register_eqa_task()
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/eqa/__init__.py", line 13, in _try_register_eqa_task
    from habitat.tasks.eqa.eqa import EQATask  # noqa: F401
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/eqa/eqa.py", line 17, in <module>
    from habitat.tasks.nav.nav import NavigationEpisode, NavigationTask
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/nav/nav.py", line 39, in <module>
    from habitat.utils.visualizations import fog_of_war, maps
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/utils/visualizations/__init__.py", line 7, in <module>
    from habitat.utils.visualizations import maps, utils
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/utils/visualizations/maps.py", line 18, in <module>
    from habitat.sims.habitat_simulator.habitat_simulator import HabitatSim
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 83, in <module>
    class HabitatSimRGBSensor(RGBSensor):
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 85, in HabitatSimRGBSensor
    sim_sensor_subtype: habitat_sim.SensorSubType
AttributeError: module 'habitat_sim' has no attribute 'SensorSubType'
```

Habitat sim version/branch : 781f787714c7f8e6da7b62c69390ada7109167ae

Habitat sim installed successfully 
[image]
[image]

erikwijmans, 2021-04-16:
The version of habitat-sim you are using isn't compatible with the version of habitat-lab you are using.  Either upgrade habitat-sim or down grade habitat-lab.  The easiest way to ensure compatibility is to have both be the same release number.

SensorSubType was introduced in https://github.com/facebookresearch/habitat-sim/pull/933 if that's helpful.

closed 2021-04-16 by erikwijmans

LaughBuddha, 2021-04-16:
Hi @erikwijmans 

Thanks for your reply.

I just did fresh clone of habitat-lab and installed the master version.
[image]

Upon running `python examples/example.py` getting the same error
```
(venv_1) sidgoel@node-1080ti-0:~/habitat-lab$ python examples/example.py 
Traceback (most recent call last):
  File "examples/example.py", line 7, in <module>
    import habitat
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/__init__.py", line 9, in <module>
    from habitat.core.benchmark import Benchmark
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/core/benchmark.py", line 18, in <module>
    from habitat.core.env import Env
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/core/env.py", line 20, in <module>
    from habitat.datasets import make_dataset
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/__init__.py", line 7, in <module>
    from habitat.datasets.registration import make_dataset  # noqa: F401 .
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/registration.py", line 23, in <module>
    _try_register_objectnavdatasetv1()
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/object_nav/__init__.py", line 20, in _try_register_objectnavdatasetv1
    from habitat.datasets.object_nav.object_nav_dataset import (  # noqa: F401
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/object_nav/object_nav_dataset.py", line 15, in <module>
    from habitat.datasets.pointnav.pointnav_dataset import (
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 15, in <module>
    from habitat.tasks.nav.nav import (
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/__init__.py", line 7, in <module>
    from habitat.tasks.registration import make_task  # noqa: F401
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/registration.py", line 24, in <module>
    _try_register_eqa_task()
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/eqa/__init__.py", line 13, in _try_register_eqa_task
    from habitat.tasks.eqa.eqa import EQATask  # noqa: F401
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/eqa/eqa.py", line 17, in <module>
    from habitat.tasks.nav.nav import NavigationEpisode, NavigationTask
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/tasks/nav/nav.py", line 39, in <module>
    from habitat.utils.visualizations import fog_of_war, maps
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/utils/visualizations/__init__.py", line 7, in <module>
    from habitat.utils.visualizations import maps, utils
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/utils/visualizations/maps.py", line 18, in <module>
    from habitat.sims.habitat_simulator.habitat_simulator import HabitatSim
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 83, in <module>
    class HabitatSimRGBSensor(RGBSensor):
  File "/mnt/beegfs/home/sidgoel/habitat-lab/habitat/sims/habitat_simulator/habitat_simulator.py", line 85, in HabitatSimRGBSensor
    sim_sensor_subtype: habitat_sim.SensorSubType
AttributeError: module 'habitat_sim' has no attribute 'SensorSubType'
```