facebookresearch/habitat-sim #1176: EGL_BAD_ACCESS on Habitat-sim headless

yuyecreus, 2021-04-17:
Hi there,

So I am working on a remote server that runs slurm over Ubuntu 18.04 and that has many GPUs with the following configuration (returned after running nvidia-smi):

Driver Version: 460.67 
CUDA Version: 11.2 

Since I am working on a remote server, I have the headless installation of Habitat-Sim.

Now, I have a strange issue that is not really consistent, as in the majority of cases, when I run any command that executes Habitat-sim there is the following problem:

Platform::WindowlessEglApplication::tryCreateContext(): cannot initialize EGL: EGL_BAD_ACCESS
WindowlessContext: Unable to create windowless context

However, in some very unusual runs, without changing anything of the command, the program is capable of accessing the GPU without any problem and runs properly. 

I have no idea why there are runs that work correctly and others that return this EGL_BAD_ACCESS. If you have any clue that would help me very much.

Thank you guys :)

erikwijmans, 2021-04-18:
Two things:

1. `export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON` will give more verbose logging about context initialization, which can be helpful
2. I have sometimes seems a similar error (I can't quite remember if it's `EGL_BAD_ACCESS`) when quickly destroying and reinitializing the simulator in subprocesses (i.e. through `habitat.VectorEnv` in checkpoint evaluation for habitat-baselines.  All seems fine for rapid destruction and initialization within the same process).  As far as I can tell that is an nvidia driver issue and I worked it by adding a `time.sleep(30)` at the end of eval for a checkpoint.

yuyecreus, 2021-04-19:
> Two things:
>
> 1. `export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON` will give more verbose logging about context initialization, which can be helpful
> 2. I have sometimes seems a similar error (I [...]

Thank you Erik. I now attach the full output of the problem:

Platform::WindowlessEglApplication: eglQueryDeviceStringEXT(EGLDevice=0): EGL_NV_device_cuda EGL_EXT_device_drm EGL_EXT_device_query_name
Platform::WindowlessEglApplication: found 9 EGL devices, choosing EGL device 0 for CUDA device 0
eglInitialize(): EGL_BAD_ACCESS error: In eglInitialize: EGLDisplay (0x55d867a2da40): Backend failed to allocate context
Platform::WindowlessEglApplication::tryCreateContext(): cannot initialize EGL: EGL_BAD_ACCESS
WindowlessContext: Unable to create windowless context

Now, it is true that the error shows through habitat.VectorEnv as when I train or eval a PPO policy. However, it also shows up when executing any command that runs habitat-sim (i.e. generate top-down-map or perform random navigation in a scene) without being related to habitat.VectorEnv in these cases.

Thank you very much for helping out!

erikwijmans, 2021-04-21:
Unfortunately this is likely a driver issue.  I'm not sure if we have any machines that are on 460.67, but a fair number are on 460.32 and that seems fine for our kernel versions/OS.

closed 2021-07-28 by yuyecreus