facebookresearch/habitat-sim #1178: No OpenGL context when using ray

Skylark0924, 2021-04-19:
## ❓ Questions and Help

Hi there, I met a similar error like  @#1038 when I try to use MBPO for PointNav with ray framework, and the solution of appointing GPU_id is not working for me. 

The log is:

```(pid=13770) I0419 16:15:22.918272 13770 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
(pid=13770) I0419 16:15:22.918316 13770 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Dataset) : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
(pid=13770) I0419 16:15:22.918406 13770 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
(pid=13770) I0419 16:15:22.918443 13770 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
(pid=13770) I0419 16:15:22.918481 13770 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
(pid=13770) I0419 16:15:22.918502 13770 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
(pid=13770) I0419 16:15:22.918514 13770 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
(pid=13770) I0419 16:15:22.918524 13770 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
(pid=13770) I0419 16:15:22.918556 13770 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
(pid=13770) I0419 16:15:22.918579 13770 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
(pid=13770) I0419 16:15:22.918593 13770 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
(pid=13770) I0419 16:15:22.918606 13770 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
(pid=13770) I0419 16:15:22.918625 13770 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
(pid=13770) I0419 16:15:22.918651 13770 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
(pid=13770) I0419 16:15:22.918658 13770 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
(pid=13770) I0419 16:15:22.919178 13770 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
(pid=13770) I0419 16:15:22.919181 13770 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
(pid=13770) I0419 16:15:22.922598 13770 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Physics Manager) : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  does not exist.
(pid=13770) I0419 16:15:22.922616 13770 PhysicsAttributesManager.cpp:27] File (./data/default.physics_config.json) not found, so new default physics manager attributes created and registered.
(pid=13770) I0419 16:15:22.922621 13770 AbstractObjectAttributesManagerBase.h:175] AbstractObjectAttributesManager<T>::createObject  (Stage) : Making attributes with handle : /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb
(pid=13770) I0419 16:15:22.922624 13770 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb changed to proposed JSON configuration filename : /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.stage_config.json
(pid=13770) I0419 16:15:22.922633 13770 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Stage) : Proposing JSON name : /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.stage_config.json from original name : /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb | This file  does not exist.
(pid=13770) I0419 16:15:22.922693 13770 AbstractObjectAttributesManagerBase.h:181] AbstractObjectAttributesManager<T>::createObject  (Stage) : Done making attributes with handle : /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb
(pid=13770) I0419 16:15:22.922695 13770 AbstractObjectAttributesManagerBase.h:188] File (/home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
(pid=13770) I0419 16:15:22.922704 13770 Simulator.cpp:156] Loading navmesh from /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.navmesh
(pid=13770) I0419 16:15:22.923107 13770 Simulator.cpp:158] Loaded.
(pid=13770) I0419 16:15:22.923120 13770 SceneGraph.h:93] Created DrawableGroup: 
(pid=13770) Renderer: GeForce RTX 2080/PCIe/SSE2 by NVIDIA Corporation
(pid=13770) OpenGL version: 4.6.0 NVIDIA 460.67
(pid=13770) Using optional features:
(pid=13770)     GL_ARB_ES2_compatibility
(pid=13770)     GL_ARB_direct_state_access
(pid=13770)     GL_ARB_get_texture_sub_image
(pid=13770)     GL_ARB_invalidate_subdata
(pid=13770)     GL_ARB_multi_bind
(pid=13770)     GL_ARB_robustness
(pid=13770)     GL_ARB_separate_shader_objects
(pid=13770)     GL_ARB_texture_filter_anisotropic
(pid=13770)     GL_ARB_texture_storage
(pid=13770)     GL_ARB_texture_storage_multisample
(pid=13770)     GL_ARB_vertex_array_object
(pid=13770)     GL_KHR_debug
(pid=13770) Using driver workarounds:
(pid=13770)     no-forward-compatible-core-context
(pid=13770)     nv-egl-incorrect-gl11-function-pointers
(pid=13770)     no-layout-qualifiers-on-old-glsl
(pid=13770)     nv-zero-context-profile-mask
(pid=13770)     nv-implementation-color-read-format-dsa-broken
(pid=13770)     nv-cubemap-inconsistent-compressed-image-size
(pid=13770)     nv-cubemap-broken-full-compressed-image-query
(pid=13770)     nv-compressed-block-size-in-bits
(pid=13770) I0419 16:15:22.966691 13770 ResourceManager.cpp:234] ResourceManager::loadStage : Not loading semantic mesh
(pid=13770) I0419 16:15:22.966703 13770 ResourceManager.cpp:262] ResourceManager::loadStage : start load render asset /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb.
(pid=13770) I0419 16:15:22.966706 13770 ResourceManager.cpp:569] ResourceManager::loadStageInternal : Attempting to load stage /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.glb 
(pid=13770) I0419 16:15:22.966722 13770 ResourceManager.cpp:1119] Importing Basis files as BC7 for dhjEzFoUFzH.glb
(pid=13770) W0419 16:15:33.799813 13770 Simulator.cpp:248] :
(pid=13770) ---
(pid=13770)  The active scene does not contain semantic annotations. 
(pid=13770) ---
(pid=13770) I0419 16:15:33.836005 13770 simulator.py:213] Loaded navmesh /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.navmesh
(pid=13770) GPU is 0
(pid=13770) I0419 16:19:08.385867 13770 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json
(pid=13770) I0419 16:19:08.385915 13770 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Dataset) : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
(pid=13770) I0419 16:19:08.385993 13770 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) crea [... cut ...]
(pid=13770) ---
(pid=13770)  The active scene does not contain semantic annotations. 
(pid=13770) ---
(pid=13770) I0419 16:19:19.259064 13770 simulator.py:213] Loaded navmesh /home/skylark/datasets/habitat/data/scene_datasets/mp3d/dhjEzFoUFzH/dhjEzFoUFzH.navmesh
(pid=13770) Simulator::gpuDevice: no OpenGL context.
```
 I have no idea why it loaded the env twice since it has already found the `OpenGL version: 4.6.0 NVIDIA 460.67` the first time. Is there any idea about how to fix it? Thanks a lot!

erikwijmans, 2021-04-19:
Are you trying to create more than one instance of the simulator in the same process?  That is currently not supported.

Skylion007, 2021-04-20:
Yeah, you should be able to use Ray, but you need to make sure it's the multiprocess VectorEnv.

Skylark0924, 2021-04-27:
@Skylion007 Does that mean I need to create an Env which inherits VectorEnv class?

Skylark0924, 2021-04-27:
@erikwijmans Actually I am just trying to use a single env, and 'NUM_PROCESSES': 1 in the config file.

Skylark0924, 2021-04-27:
I finally found that it caused by the evaluation env construction since MBPO creates an independent evaluation environment while creating a training environment. By commenting out the evaluation environment, codes the issue is solved.

closed 2021-04-27 by Skylion007