facebookresearch/habitat-sim #1209: Problem installing Habitat-Sim

leokonto, 2021-05-01:
**System Settings:**
Ubuntu 18.04 LTS
Conda 4.9.2

**Execution order**
git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim
conda create -n habitat python=3.6 cmake=3.14.0
conda activate habitat
pip install -r requirements.txt
sudo apt-get update || true
sudo apt-get install -y --no-install-recommends \
     libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils xorg-dev freeglut3-dev
python setup.py install

-----------------------------------------------------------------------------------------------------------------------------

After I run the last command I get following output:

running install
running bdist_egg
running egg_info
creating habitat_sim.egg-info
writing habitat_sim.egg-info/PKG-INFO
writing dependency_links to habitat_sim.egg-info/dependency_links.txt
writing requirements to habitat_sim.egg-info/requires.txt
writing top-level names to habitat_sim.egg-info/top_level.txt
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest file 'habitat_sim.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'habitat_sim.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/examples
copying examples/demo_runner.py -> build/lib.linux-x86_64-3.6/examples
copying examples/ab_test.py -> build/lib.linux-x86_64-3.6/examples
copying examples/__init__.py -> build/lib.linux-x86_64-3.6/examples
copying examples/benchmark.py -> build/lib.linux-x86_64-3.6/examples
copying examples/settings.py -> build/lib.linux-x86_64-3.6/examples
copying examples/example.py -> build/lib.linux-x86_64-3.6/examples
creating build/lib.linux-x86_64-3.6/tests
copying tests/test_controls.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_random_seed.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_profiling_utils.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_examples.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_utils.py -> build/lib.linux-x86_64-3.6/tests
copying tests/__init__.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_physics.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_greedy_follower.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_light_setup.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_gfx.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_simulator.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_agent.py -> build/lib.linux-x86_64-3.6/tests
copying tests/conftest.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_configs.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_sensors.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_attributes_managers.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_snap_point.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_navmesh.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_data_extraction.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_semantic_scene.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_noise_models.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_compare_profiles.py -> build/lib.linux-x86_64-3.6/tests
copying tests/test_pyrobot_noisy_controls.py -> build/lib.linux-x86_64-3.6/tests
creating build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/scene.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/attributes.py -> build/lib.linux-x86_64-3.6/habitat_sim
copying habitat_sim/errors.py -> build/lib.linux-x86_64-3.6/habitat_sim
[... 150 lines cut ...]
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 204, in run
    self.build_extension(ext)
  File "setup.py", line 273, in build_extension
    env=env,
  File "/home/leo/anaconda3/envs/habitat/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/home/leo/habitat/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/leo/habitat/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/leo/anaconda3/envs/habitat/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-GNinja', '-DBUILD_GUI_VIEWERS=ON', '-DTARGET_HEADLESS=OFF', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF', '-DBUILD_WITH_CUDA=OFF']' returned non-zero exit status 1.

erikwijmans, 2021-05-01:
```
Could not find compiler set in environment variable CC:

/home/leo/anaconda3/envs/habitat/bin/x86_64-conda_cos6-linux-gnu-cc.
Call Stack (most recent call first):
CMakeLists.txt:15 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
```

Seems like your system does not have a c/cpp compiler installed.  `sudo apt install build-essential` will get you one

leokonto, 2021-05-01:
A simple reboot and sudo apt-get upgrade seemed to do the trick actually. Still thank you for your timely answer!

closed 2021-05-01 by leokonto