facebookresearch/habitat-sim #1232: .glb files of gibson dataset is black in habitat-viewer

glattm, 2021-05-12:
I wanted to try out the gibson dataset with habitat. I downloaded it from https://github.com/StanfordVL/GibsonEnv#database
and watch it with the habitat viewer:

`habitat-viewer /path/to/Allensville.glb`

Every file is only displayed as a black  surface. I'm able to navigate in them with arrow keys or setting the agent in random places.

What am i doing wrong ?

aclegg3, 2021-05-18:
Loading gibson's Allensville scene with the viewer via (`./build/viewer -- /path/to/gibson/Allensville.glb`) I see:
[image]

glattm, 2021-05-19:
When i try to run `./build/viewer -- /path/to/gibson/Allensville.glb` i get the following screen
[image]
I can navigate and print the agents position onto the screen, so it seems to work but only with a black screen

The only file that works for me is "Ackermanville.glb"
If i run `./build/viewer -- /path/to/gibson/Ackermanville.glb` i get this screen:
[image]

which seems to work fine

closed 2021-08-16 by dhruvbatra