facebookresearch/habitat-sim #1245: Light tutorial not working

gianscarpe, 2021-05-18:
## Habitat-Sim version
v 0.1.7

## 🐛 Bug

Launching the `lightning_tutorial` I get
```
  File "examples/tutorials/lighting_tutorial.py", line 102, in main
    LightInfo(vector=[0.0, 2.0, 0.6, 0.0], model=LightPositionModel.Global)
AttributeError: type object 'habitat_sim._ext.habitat_sim_bindings.LightPositio' has no attribute 'Global'
```
Is this a unknown issue? It seems like a type in the C++ bindings 

## Steps to Reproduce

Steps to reproduce the behavior:

Using habitat 0.1.7 (installed with conda)
`python examples/tutorials/lighting_tutorial.py`
 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior
I expect it to work as in https://aihabitat.org/docs/habitat-sim/lighting-setups.html

## Additional context

aclegg3, 2021-05-18:
This tutorial is working for me on master branch. 

Did you install from an old conda build like stable v0.1.7 and then run the script on master? That could be the problem. You can try building master from source, conda installing the nightly build, or cloning the repo at v0.1.7 to get scripts compatible with that version.

gianscarpe, 2021-05-20:
This solved my issue @aclegg3 ! Thank you :)

closed 2021-05-20 by gianscarpe