facebookresearch/habitat-sim #1251: Can't run on headless machine (EGL and magnum errors)

sparisi, 2021-05-19:
Hi, I am trying to run Habitat `RLEnv` with [this algorithm](https://github.com/facebookresearch/impact-driven-exploration) (also from facebookresearch) on a headless machine, but I am getting two errors.

First, I clone both `habitat-lab` and `habitat-sim` stable branches.
I follow the instructions and install `habitat-lab` with `pip` and `habitat-sim` with conda and the headless option. No errors at this point. I can also run `python examples/examples.py` successfully.
However, if I try to run the same environment with the RL code I linked I get 

```
I0519 14:27:53.016191 2270664 Simulator.cpp:158] Loaded.
I0519 14:27:53.016216 2270664 SceneGraph.h:93] Created DrawableGroup:
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context
```

According to [the common issues section](https://github.com/facebookresearch/habitat-sim/blob/master/BUILD_FROM_SOURCE.md#common-build-issues), I have to build from source with `DEGL_LIBRARY`. I do it, but with one difference: I don't have `/usr/lib/x86_64-linux-gnu/nvidia-opengl/libEGL.so` but `/usr/lib/x86_64-linux-gnu/nvidia-opengl/libEGL_nvidia.so.0`

I build from source (no errors) and then run the same code with `LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/nvidia-opengl:${LD_LIBRARY_PATH}`, but this time I get another error. If I try to run the example I get

```
Traceback (most recent call last):
  File "examples/example.py", line 10, in <module>
    import demo_runner as dr
  File "/private/home/sparisi/habitat-baselines/habitat-sim/examples/demo_runner.py", line 15, in <module>
    from settings import default_sim_settings, make_cfg
  File "/private/home/sparisi/habitat-baselines/habitat-sim/examples/settings.py", line 5, in <module>
    import habitat_sim
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/agent/__init__.py", line 9, in <module>
    from .agent import *  # noqa: F401,F403
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 14, in <module>
    from habitat_sim import bindings as hsim
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/bindings/__init__.py", line 10, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: generic_type: type "SceneNode" referenced unknown base type "Magnum::SceneGraph::Object<Magnum::SceneGraph::BasicTranslationRotationScalingTransformation3D<float> >"
```

If I try the other code I get

```
Traceback (most recent call last):
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/private/home/sparisi/habitat-baselines/minigrid/src/utils.py", line 345, in act
    raise e
  File "/private/home/sparisi/habitat-baselines/minigrid/src/utils.py", line 192, in act
    gym_envs = create_env(flags)
  File "/private/home/sparisi/habitat-baselines/minigrid/src/utils.py", line 99, in create_env
    envs.append(make_habitat())
  File "/private/home/sparisi/habitat-baselines/minigrid/src/utils.py", line 57, in make_habitat
    from habitat_baselines.common.environments import NavRLEnv
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat_baselines/__init__.py", line 12, in <module>
    from habitat_baselines.il.trainers.pacman_trainer import PACMANTrainer
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat_baselines/il/trainers/pacman_trainer.py", line 24, in <module>
    from habitat_baselines.il.data.nav_data import NavDataset
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat_baselines/il/data/nav_data.py", line 15, in <module>
    from habitat.datasets.utils import VocabDict
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat/datasets/utils.py", line 19, in <module>
    from habitat.sims.habitat_simulator.actions import HabitatSimActions
  File "/private/home/sparisi/habitat-baselines/habitat-lab/habitat/sims/habitat_simulator/actions.py", line 12, in <module>
    import habitat_sim
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/agent/__init__.py", line 9, in <module>
    from .agent import *  # noqa: F401,F403
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/agent/agent.py", line 14, in <module>
    from habitat_sim import bindings as hsim
  File "/private/home/sparisi/.conda/envs/hab/lib/python3.7/site-packages/habitat_sim/bindings/__init__.py", line 10, in <module>
    from habitat_sim._ext.habitat_sim_bindings import (
ImportError: generic_type: type "MapStringString" is already registered!
```

I have seen that this error have been reported before and it is due to `magnum`, but I cannot find a fix.

Any help is appreciated, thanks!

erikwijmans, 2021-05-19:
You do not build from source, that `-DEGL_LIBRARY` flag is only needed when cmake isn't finding libegl, which isn't a thing when grabbing conda binaries.  It's an issue when building from source, not when grabbing binaries.

Your current issue is likely that magnum and habitat-sim are out of sync.  Likely just forcing a reinstall of conda binaries will fix that.

As for your original issue, as you are able to run the example, my guess is that the code you are adapting initializes CUDA before the RLEnv.  It's a known issue with CUDA and EGL that EGL needs to be initialized first if you want OpenGL and CUDA in the same process.  Using `habitat.VectorEnv` is another way to get around that as that creates fresh processes.

sparisi, 2021-05-20:
Thanks, `VectorEnv` works!

closed 2021-05-20 by erikwijmans