facebookresearch/habitat-sim #1278: depth to world coordinate

zhangfuyang, 2021-05-28:
## ❓ Questions and Help
Hi, I am working on using a depth sensor to get each pixel's world coordinate. I modified from [this](https://aihabitat.org/docs/habitat-lab/view-transform-warp.html) tutorial, however, the result seems not correct.

Here is my code:

```
K = np.array([
    [1/np.tan(hfov / 2.), 0., 0., 0.],
    [0., 1/np.tan(hfov / 2.), 0., 0.],
    [0., 0., 1., 0.],
    [0., 0., 0., 1.]
])
xs, ys = np.meshgrid(np.linspace(-1,1,W), np.linspace(1,-1,W))
depth = obs['depth'][:,:,0].reshape(1,W,W)*10
xs = xs.reshape(1,W,W)
ys = ys.reshape(1,W,W)

xys = np.vstack((xs * depth , ys * depth, -depth, np.ones(depth.shape)))
xys = xys.reshape(4, -1)
xy_c0 = np.matmul(np.linalg.inv(K), xys)

quaternion_ = sensor_states['depth'].rotation
translation_ = sensor_states['depth'].position
rotation_ = quaternion.as_rotation_matrix(quaternion_)
T_world_camera = np.eye(4)
T_world_camera[0:3,0:3] = rotation_
T_world_camera[0:3,3] = translation_
print(T_world_camera)

xyz = np.matmul(T_world_camera, xy_c0)
xyz = xyz / xyz[3,:]
```

How do I tell this is not correct? Here is one example I tested.
Left is RGB image, right is depth image. The red cylinder is an object added to the scene. Obviously, the cylinder is the nearest object to the camera.
[image]
Below is the distance of each pixel to agent location in world coordinate, left is original distance image, right is the one I mask the location of top 5000 smallest distance.
[image]
As you can see, the correct result should be masking all around the cylinder. However, it's not. 

So there must be something wrong with the code. If you have any clue please help me. Thank you guys!

erikwijmans, 2021-05-28:
```
depth = obs['depth'][:,:,0].reshape(1,W,W)*10
```
Since you are posting here, I assume you are not using habitat-lab?  If so, you don't need to multiply by 10 as the depth normalization happens in habitat-lab, habitat-sim directly outputs metric depth.

zhangfuyang, 2021-05-29:
Thanks! I found the stupid bug I wrote :P

closed 2021-05-29 by zhangfuyang

arjung128, 2022-02-19:
@zhangfuyang could you please share what the bug was? Do you have a function which gives each pixel's world coordinate? Thanks!

zhangfuyang, 2022-02-19:
I remember the code above is bug-free, which gives you each pixel's world coordinate. Just to make sure hfov should be in radian [0, 2pi], not in [0, 360]. That was the stupid bug I made :)

derrick-xwp, 2022-06-28:
> ```
> depth = obs['depth'][:,:,0].reshape(1,W,W)*10
> ```
> 
> Since you are posting here, I assume you are not using habitat-lab? If so, you don't need to multiply by 10 as the depth normalization happens in habitat-lab, habitat-sim directly outputs metric depth.

How to get the bounding box of the scene in world coordinate?