facebookresearch/habitat-sim #1285: get_topdown_map returns the same map for different environments

ruddradev, 2021-05-31:
## Habitat-Sim version
v0.1.7

## Habitat-Lab version
v0.1.7

Hello All,

I am facing issue while trying to iterate over multiple environments and trying to dump their top down maps. 

```
for i in range(len(env.episodes)):
        env_name = str(i)
        print("Environment Name = ", env_name)
        if(not os.path.exists(dump_path+env_name)):
            os.makedirs(dump_path+env_name)
        observations = env.reset()
        # print(env.episodes[i])
        meters_per_pixel = 0.01
        height = 0.1
        simulator = env.sim
        # simulator.reset()
        top_down_map = maps.get_topdown_map(env.sim.pathfinder, height, meters_per_pixel=meters_per_pixel)
```
When I display the get_topdown_map, it gives me the same map, however in the console when I check the logs, it is showing a different glb file being loaded for every episode. Though I am using Habitat lab for the iterations, I felt it was appropriate to ask here since topdown_map is a function of habitat_sim. 

Any pointers in the right direction will be great.

erikwijmans, 2021-06-01:
Are the episodes actually loading different scenes?  You can check the scene for an episode with `envcurrent_episode.scene_id`

ruddradev, 2021-06-02:
Hello,
Yes, I just confirmed the name of the scenes and they are different barring the first two. But the maps are always coming out as same. 

Here are the logs of first 5 episodes - 

```
Environment Name =  0
Scene_ID =  /home/ruddra/Dataset/Gibson/habitat/scene_datasets/gibson/Greigsville.glb
Replacing relative pose file
Replacing absolute pose file
RGB folder has old files, removing them..
Depth folder has old files, removing them..
Enter an instruction: 0
stop
Delta pose =  (-0.0, 0.0, 0.0)
Point Goal =  [ 8.723369  -2.6321084]
Current Position =  (2.90618, 4.69643, -2.3325732854223213)
Environment Name =  1
Scene_ID =  /home/ruddra/Dataset/Gibson/habitat/scene_datasets/gibson/Greigsville.glb
Replacing relative pose file
Replacing absolute pose file
RGB folder has old files, removing them..
Depth folder has old files, removing them..
Enter an instruction: 0
stop
Delta pose =  (-0.0, 0.0, 0.0)
Point Goal =  [11.3213415 -2.4535081]
Current Position =  (1.17589, 0.27251, -2.5232260127825255)
Environment Name =  2
Scene_ID =  /home/ruddra/Dataset/Gibson/habitat/scene_datasets/gibson/Swormville.glb
Replacing relative pose file
Replacing absolute pose file
RGB folder has old files, removing them..
Depth folder has old files, removing them..
Enter an instruction: 0
stop
Delta pose =  (-0.0, 0.0, 0.0)
Point Goal =  [5.446701  1.8531313]
Current Position =  (0.08615, 3.92178, -3.0546248417813406)
Environment Name =  3
Scene_ID =  /home/ruddra/Dataset/Gibson/habitat/scene_datasets/gibson/Scioto.glb
Created rgb dump directory
Created depth dump directory
Enter an instruction: 0
stop
Delta pose =  (-0.0, 0.0, 0.0)
Point Goal =  [7.629368   0.33910885]
Current Position =  (0.02986, 7.23818, -1.7446515187768767)
Environment Name =  4
Scene_ID =  /home/ruddra/Dataset/Gibson/habitat/scene_datasets/gibson/Eastville.glb
Created rgb dump directory
Created depth dump directory
Enter an instruction: 0
stop
Delta pose =  (-0.0, -0.0, 0.0)
Point Goal =  [ 3.512356 -1.695817]
Current Position =  (3.34582, 7.53066, 2.9555650176481727)
```
And here are the maps - 
[Maps.zip](https://github.com/facebookresearch/habitat-sim/files/6582420/Maps.zip)

erikwijmans, 2021-06-02:
```
env.episodes = random.sample(env.episodes, 5)
```
Things aren't really setup to have that work.  Modifying the episode list won't change the episode iterator, which is where episodes are selected from.  The episodes aren't iterated in a list, that is why `env.current_episode` exists.  If you want to select a subset of the episodes, use `config.ENVIRONMENT.ITERATOR_OPTIONS.NUM_EPISODE_SAMPLE = 5`