facebookresearch/habitat-sim #1293: Creating two simulator instances  generates the error: "Simulator::gpuDevice: no OpenGL context"

bekeronur, 2021-06-04:
## Habitat-Sim version
v0.1.7

## ❓ Questions and Help
Hi, 

I cannot create two habitat simulator instances inside the same python script, attempting to do so generates the error message: "Simulator::gpuDevice: no OpenGL context", and the program then crashes. The issue persists if the two simulator instances are assigned different GPU ID's.

There is no issue if I run two identical scripts in parallel as two separate programs (or if I run the commands interactively across two parallel python interpreters on different terminal instances), and nvidia-smi shows that the allocated GPU memory indeed corresponds to two simulator instances for this case. 

Thanks a lot in advance for the help!

## Additional Context
I implemented a gym environment that maintains a habitat-sim object as one of its fields, together with some additional functionality implemented around it for reward assignment, episode initialization, curriculum management, etc. I need to parallelize multiple such environments, and that is when I first encountered the issue. The issue can be reproduced without any of these additions, by simply creating two habitat-sim instances.

## Code to Reproduce the Issue
The code below creates a simple sim configuration, and instantiates two simulators.
```python
import os
import math
import habitat_sim

if __name__ == "__main__":
    
    # Set scene path
    SCENE_PATH = "./habitat-lab/data/scene_datasets/habitat-test-scenes/apartment_1.glb"
    
    # Setup general parameters (scene path, gpu, physics)
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.gpu_device_id = 0
    sim_cfg.scene_id = SCENE_PATH
    sim_cfg.enable_physics = False

    # Sensor configuration
    sensor_specs = []
    color_sensor_1st_person_spec = habitat_sim.CameraSensorSpec()
    color_sensor_1st_person_spec.uuid = "color_sensor_1st_person"
    color_sensor_1st_person_spec.sensor_type = habitat_sim.SensorType.COLOR
    color_sensor_1st_person_spec.resolution = [544, 720]
    color_sensor_1st_person_spec.position = [0.0, 1, 0.5] 
    color_sensor_1st_person_spec.orientation = [(-math.pi / 8.0),0.0,0.0,]
    color_sensor_1st_person_spec.sensor_subtype = habitat_sim.SensorSubType.PINHOLE
    sensor_specs.append(color_sensor_1st_person_spec)

    # Agent configuration
    agent_cfg = habitat_sim.agent.AgentConfiguration()
    agent_cfg.sensor_specifications = sensor_specs
    agent_cfg.action_space = {
        "move_forward": habitat_sim.agent.ActionSpec(
            "move_forward", habitat_sim.agent.ActuationSpec(amount=0.25) # default = 0.25
        ),
        "turn_left": habitat_sim.agent.ActionSpec(
            "turn_left", habitat_sim.agent.ActuationSpec(amount=30.0) # default = 0.25
        ),
        "turn_right": habitat_sim.agent.ActionSpec(
            "turn_right", habitat_sim.agent.ActuationSpec(amount=30.0) # default = 0.25
        ),

    }

    cfg = habitat_sim.Configuration(sim_cfg, [agent_cfg])

    # Create 2 simulator instances
    print('Creating sim1...')
    sim1 = habitat_sim.Simulator(cfg)
    print('Creating sim2...')
    sim2 = habitat_sim.Simulator(cfg)
```

## Outputs
```
Creating sim1...
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0604 14:19:06.148123 16987 SemanticScene.h:155] SemanticScene::loadSemanticSceneDescriptor : File ./habitat-lab/data/scene_datasets/habitat-test-scenes/apartment_1.scn does not exist.  Aborting load.
Creating sim2...
E0604 14:19:07.502121 16987 SemanticScene.h:155] SemanticScene::loadSemanticSceneDescriptor : File ./habitat-lab/data/scene_datasets/habitat-test-scenes/apartment_1.scn does not exist.  Aborting load.
Simulator::gpuDevice: no OpenGL context.
Aborted (core dumped)
```

erikwijmans, 2021-06-04:
Opening two instances of the simulator is not supported

bekeronur, 2021-06-04:
I see, thanks.

closed 2021-06-04 by bekeronur