facebookresearch/habitat-sim #1298: Platform::WindowlessEglApplication::tryCreateContext(): cannot initialize EGL: EGL_NOT_INITIALIZED

yifan123, 2021-06-07:
## ❓ Questions and Help

Hi there,

I'm trying to set up habitat-sim on a remote CentOS Linux release 7.4.1708 (Core) machine and running into the following error when trying to run simulations. Any help with installation would be appreciated:

    I0607 19:03:14.663993 106203 SceneGraph.h:93] Created 
    DrawableGroup:Platform::WindowlessEglApplication::tryCreateContext(): cannot initialize EGL: EGL_NOT_INITIALIZED
    WindowlessContext: Unable to create windowless context

## Specs
CentOS Linux release 7.4.1708 (Core) with 8 NVIDIA GPUs on slurm.

erikwijmans, 2021-06-07:
Please take a look at https://github.com/facebookresearch/habitat-sim/issues/288.  Also, running with `export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON` will increase logging verbosity for EGL which may help.

yifan123, 2021-06-08:
After running with export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON, 
```
sim_cfg.physics_config_file = ./data/default.physics_config.json
==== Initialized Sensor Spec: =====
Sensor uuid:  color_sensor
Sensor type:  SensorType.COLOR
Sensor position:  [0.  1.5 0. ]
===================================
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0608 13:24:00.117231 285688 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : default changed to proposed JSON configuration filename : default.scene_dataset_config.json                                           I0608 13:24:00.117431 285688 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Dataset) : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I0608 13:24:00.132011 285688 AssetAttributesManager.cpp:117] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered. I0608 13:24:00.132084 285688 AssetAttributesManager.cpp:117] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.                                     I0608 13:24:00.132143 285688 AssetAttributesManager.cpp:117] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.         I0608 13:24:00.132980 285688 AssetAttributesManager.cpp:117] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0608 13:24:00.133015 285688 AssetAttributesManager.cpp:117] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0608 13:24:00.133033 285688 AssetAttributesManager.cpp:117] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0608 13:24:00.133095 285688 AssetAttributesManager.cpp:117] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0608 13:24:00.133136 285688 AssetAttributesManager.cpp:117] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0608 13:24:00.133163 285688 AssetAttributesManager.cpp:117] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0608 13:24:00.133193 285688 AssetAttributesManager.cpp:117] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0608 13:24:00.133225 285688 AssetAttributesManager.cpp:117] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0608 13:24:00.133255 285688 AssetAttributesManager.cpp:117] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered. 
I0608 13:24:00.133268 285688 AssetAttributesManager.cpp:105] AssetAttributesManager::buildCtorFuncPtrMaps : Built default primitive asset templates : 12
I0608 13:24:00.134629 285688 SceneDatasetAttributesManager.cpp:23] File (default) not found, so new default dataset attributes created and registered.
I0608 13:24:00.134645 285688 MetadataMediator.cpp:47] MetadataMediator::createDataset : Dataset default successfully created.
I0608 13:24:00.139792 285688 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Physics Manager) : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I0608 13:24:00.140772 285688 PhysicsAttributesManager.cpp:27] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I0608 13:24:00.140791 285688 AbstractObjectAttributesManagerBase.h:175] AbstractObjectAttributesManager<T>::createObject  (Stage) : Making attributes with handle : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
I0608 13:24:00.140807 285688 ManagedContainerBase.cpp:19] ManagedContainerBase::convertFilenameToJSON : Filename : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb changed to proposed JSON configuration filename : data/scene_datasets/habitat-test-scenes/skokloster-castle.stage_config.json
I0608 13:24:00.140949 285688 AttributesManagerBase.h:283] AttributesManager<T>::createFromJsonOrDefaultInternal  (Stage) : Proposing JSON name : data/scene_datasets/habitat-test-scenes/skokloster-castle.stage_config.json from original name : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb | This file  does not exist.
I0608 13:24:00.141317 285688 AbstractObjectAttributesManagerBase.h:181] AbstractObjectAttributesManager<T>::createObject  (Stage) : Done making attributes with handle : data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
I0608 13:24:00.141330 285688 AbstractObjectAttributesManagerBase.h:188] File (data/scene_datasets/habitat-test-scenes/skokloster-castle.glb) exists but is not a recognized config filename extension, so new default Stage attributes created and registered.
I0608 13:24:00.141350 285688 Simulator.cpp:156] Loading navmesh from data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
I0608 13:24:00.143280 285688 Simulator.cpp:158] Loaded.
I0608 13:24:00.143309 285688 SceneGraph.h:93] Created DrawableGroup:
Platform::WindowlessEglApplication::tryCreateContext(): cannot initialize EGL: EGL_NOT_INITIALIZED
WindowlessContext: Unable to create windowless context 
```
Also when run `ldconfig -p | grep EGL`, I get 
```
libEGL_mesa.so.0 (libc6,x86-64) => /usr/lib64/libEGL_mesa.so.0
libEGL.so.1 (libc6,x86-64) => /usr/lib64/libEGL.so.1
libEGL.so (libc6,x86-64) => /usr/lib64/libEGL.so
```
Is it because I lack `libEGL_nvidia.so` ?
If so, what should I install? thank you

closed 2021-08-16 by dhruvbatra