facebookresearch/habitat-sim #1315: Question on the position of sensor

AgentEXPL, 2021-06-15:
The following code is from master/examples/tutorials/stereo_agent.py  (https://github.com/facebookresearch/habitat-sim/blob/master/examples/tutorials/stereo_agent.py#L29), I'm sorry that I could not understand why to use 1.5 * habitat_sim.geo.UP rather than 1.5 + habitat_sim.geo.UP, i.e., why to use multiplication rather than addition?
----------------------------------------------------------------------------------------------------------------------
    # The left RGB sensor will be 1.5 meters off the ground
    # and 0.25 meters to the left of the center of the agent
    left_rgb_sensor.position = 1.5 * habitat_sim.geo.UP + 0.25 * habitat_sim.geo.LEFT
---------------------------------------------------------------------------------------------------------------------
Hope some good guy to provide the reason.

aclegg3, 2021-06-15:
Hey @AgentEXPL,

`habitat_sim.geo.UP` is a unit vector (i.e., length 1), so to move the camera some distance (1.5 meters) along that unit direction vector, we multiply the unit direction by the distance. 

On the other hand, adding 1.5 (a single scalar) to the vector UP would not be a valid operation.

AgentEXPL, 2021-06-16:
Hi @aclegg3 ,
Thanks a lot for your explanation. Now I understand it clearly.

closed 2021-06-16 by AgentEXPL