facebookresearch/habitat-sim #1370: AttributeError: 'Simulator' object has no attribute 'get_rigid_object_manager'

gtatiya, 2021-07-01:
## Habitat-Sim version
v0.1.7, GitHub commit: 856d4b08c1a2632626bf0d205bf46471a99502b7

## ❓ Questions and Help
I am going through this tutorial: https://github.com/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb, and when I run this notebook, I get this error:
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-7e13fe8f66e1> in <module>
      4 sim_settings = make_default_settings()
      5 # set globals: sim,
----> 6 make_simulator_from_settings(sim_settings)

<ipython-input-2-287e3109976f> in make_simulator_from_settings(sim_settings)
    130     stage_attr_mgr = sim.get_stage_template_manager()
    131     # Manager providing access to rigid objects
--> 132     rigid_obj_mgr = sim.get_rigid_object_manager()

AttributeError: 'Simulator' object has no attribute 'get_rigid_object_manager'
```

Could you please help me fix this?

jturner65, 2021-07-01:
Hello
I'm sorry that you have experienced this issue.  That colab is not compatible with that version of Habitat - that colab is from master, whereas the tagged code you are using is from 6 months ago.

If you need to use that version of Habitat, check out the older version of that colab : 
https://github.com/facebookresearch/habitat-sim/blob/v0.1.7/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb

gtatiya, 2021-07-02:
That notebook is working. Thanks!

closed 2021-07-02 by gtatiya