facebookresearch/habitat-sim #1371: Angular velocity unit?

ericchen321, 2021-07-02:
## Habitat-Sim version
Master branch, commit 5cb10 (May 2021)

## Docs and Tutorials
Did you read the docs? Yes

Did you check out the tutorials? Yes

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hello, may I ask what should be the unit for angular velocity when using `habitat_sim.physics.VelocityControl.angular_velocity`? I'm assuming the unit to be degrees / sec. But if I want an agent to rotate by 10 degrees to the left, and I set the agent to rotate at the speed of `[0, 10, 0]` for one second,  the rotation would overshoot by a lot. So I thought maybe my assumption about the unit was wrong.

The way I let the agent to rotate in pseudocode is:

```
agent_object.velocity_control.angular_velocity = np.array([0, 10, 0])
for i in range(0, 60):
  sim.step_physics(1.0/60.0)
```

aclegg3, 2021-07-06:
We use radians for most angles in Habitat-sim. This angular velocity is represented in radians/second.

ericchen321, 2021-07-07:
Thanks Alex!

closed 2021-07-08 by erikwijmans