facebookresearch/habitat-sim #1377: Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0

JiayunjieJYJ, 2021-07-09:
## Habitat-Sim version
v0.2.0
## Habitat-lab version
v0.2.0

## About my laptop
I dont have CUDA on my laptop (only CPU), the system is Ubuntu 18.04.5 on virtual machine, I create a conda env "habitat" to install habitat-lab and habit-sim.  

I install habitat-lab and habit-sim following the steps of : https://github.com/facebookresearch/habitat-lab#installation, but at the last step: python example/example.py, error occured:
```
(habitat) jiayunjie@jiayunjie-virtual-machine:~/robot_sim/habitat-lab$ python examples/example.py 
2021-07-09 09:29:01,460 Initializing dataset RearrangeDataset-v0
2021-07-09 09:29:04,901 initializing sim RearrangeSim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0709 09:29:04.926504 45316 ManagedFileBasedContainer.h:210] <Dataset>::convertFilenameToPassedExt : Filename : default changed to proposed scene_dataset_config.json filename : default.scene_dataset_config.json
I0709 09:29:04.926717 45316 AttributesManagerBase.h:365] <Dataset>::createFromJsonOrDefaultInternal : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I0709 09:29:04.932118 45316 AssetAttributesManager.cpp:120] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0709 09:29:04.933409 45316 AssetAttributesManager.cpp:120] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0709 09:29:04.933591 45316 AssetAttributesManager.cpp:120] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0709 09:29:04.933856 45316 AssetAttributesManager.cpp:120] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0709 09:29:04.934026 45316 AssetAttributesManager.cpp:120] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0709 09:29:04.939039 45316 AssetAttributesManager.cpp:120] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0709 09:29:04.939257 45316 AssetAttributesManager.cpp:120] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0709 09:29:04.939378 45316 AssetAttributesManager.cpp:120] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0709 09:29:04.939489 45316 AssetAttributesManager.cpp:120] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I0709 09:29:04.939576 45316 AssetAttributesManager.cpp:120] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I0709 09:29:04.939697 45316 AssetAttributesManager.cpp:120] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I0709 09:29:04.939776 45316 AssetAttributesManager.cpp:120] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I0709 09:29:04.939822 45316 AssetAttributesManager.cpp:108] ::constructor : Built default primitive asset templates : 12
I0709 09:29:04.952062 45316 SceneDatasetAttributesManager.cpp:36] File (default) not found, so new default dataset attributes created and registered.
I0709 09:29:04.952132 45316 MetadataMediator.cpp:127] ::createSceneDataset : Dataset default successfully created.
I0709 09:29:04.952152 45316 AttributesManagerBase.h:365] <Physics Manager>::createFromJsonOrDefaultInternal : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I0709 09:29:04.966439 45316 PhysicsAttributesManager.cpp:26] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I0709 09:29:04.966534 45316 MetadataMediator.cpp:212] ::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
I0709 09:29:04.966543 45316 MetadataMediator.cpp:183] ::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
I0709 09:29:04.966563 45316 MetadataMediator.cpp:68] ::setSimulatorConfiguration : Set new simulator config for scene/stage : data/replica_cad/stages/Stage_v3_sc1_staging.glb and dataset : default which is currently active dataset.
Platform::WindowlessEglApplication::tryCreateContext(): unable to find EGL device for CUDA device 0
WindowlessContext: Unable to create windowless context

```

## ❓ Questions and Help
I dont know how to do with the problem above, I would appreciate it if you give me useful suggestions, thanks a lot!

erikwijmans, 2021-07-09:
We don't support headless mode for non-cuda systems.  You can try a headed install, via
```shell
conda remove habitat-sim headless
conda install habitat-sim -c conda-forge -c aihabitat
```
which should work on a laptop.

JiayunjieJYJ, 2021-07-09:
> We don't support headless mode for non-cuda systems. You can try a headed install, via
> 
> ```shell
> conda remove habitat-sim headless
> conda install habitat-sim -c conda-forge -c aihabitat
> ```
> 
> which should work on a laptop.

@erikwijmans 
Thanks for your reply, I use your two commands
```
conda remove habitat-sim headless
conda install habitat-sim -c conda-forge -c aihabitat
```
and then `python example/example.py`, but another error occurs:
```
(habitat) jiayunjie@jiayunjie-virtual-machine:~/robot_sim/habitat-lab$ python examples/example.py 
2021-07-09 09:46:38,876 Initializing dataset RearrangeDataset-v0
2021-07-09 09:46:41,045 initializing sim RearrangeSim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0709 09:46:41.052733 78348 ManagedFileBasedContainer.h:210] <Dataset>::convertFilenameToPassedExt : Filename : default changed to proposed scene_dataset_config.json filename : default.scene_dataset_config.json
I0709 09:46:41.052810 78348 AttributesManagerBase.h:365] <Dataset>::createFromJsonOrDefaultInternal : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I0709 09:46:41.053840 78348 AssetAttributesManager.cpp:120] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I0709 09:46:41.053958 78348 AssetAttributesManager.cpp:120] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I0709 09:46:41.054041 78348 AssetAttributesManager.cpp:120] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I0709 09:46:41.054103 78348 AssetAttributesManager.cpp:120] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I0709 09:46:41.054145 78348 AssetAttributesManager.cpp:120] Asset attributes (cubeSolid : cubeSolid) created and registered.
I0709 09:46:41.054983 78348 AssetAttributesManager.cpp:120] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I0709 09:46:41.055083 78348 AssetAttributesManager.cpp:120] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I0709 09:46:41.055153 78348 AssetAttributesManager.cpp:120] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I0709 09:46:41.055200 78348 AssetAttributesManager.cpp:120] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
......(many lines).....
I0709 09:46:45.298972 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for laser_link.STL
I0709 09:46:45.317636 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for r_gripper_finger_link.STL
I0709 09:46:45.320147 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for r_wheel_link.STL
I0709 09:46:45.325839 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for shoulder_lift_link.dae
I0709 09:46:45.454304 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for shoulder_pan_link.dae
I0709 09:46:45.604724 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for torso_fixed_link.dae
I0709 09:46:45.615270 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for torso_lift_link.dae
I0709 09:46:45.946434 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for upperarm_roll_link.dae
I0709 09:46:46.012953 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for wrist_flex_link.dae
I0709 09:46:46.074559 78348 ResourceManager.cpp:1254] Importing Basis files as ASTC 4x4 for wrist_roll_link.dae
BulletPhysicsManager::addArticulatedObjectFromURDF: simpleObjectHandle :  hab_fetch
BulletPhysicsManager::addArticulatedObjectFromURDF: newArtObjectHandle :  hab_fetch_:0000
BulletArticulatedObject::createMotorsForAllDofs():  {(27, 24), (26, 23), (25, 21), (24, 20), (23, 19), (22, 18), (21, 17), (20, 16), (19, 15), (18, 9), (17, 8), (16, 6), (15, 4), (14, 2)}
Agent acting inside environment.
GL::Shader::compile(): compilation of vertex shader 1 failed with the following message:
0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES
GL::Shader::compile(): compilation of fragment shader 2 failed with the following message:
0:1(10): error: GLSL 4.10 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.50, 3.30, 1.00 ES, and 3.00 ES
Assertion Mn::GL::Shader::compile({vert, frag}) failed at /opt/conda/envs/py36/conda-bld/habitat-sim_1625040415038/work/src/esp/gfx/PbrShader.cpp:128
Aborted (core dumped)

```

Can you please find the reason about it? Thanks!

erikwijmans, 2021-07-09:
Looks like your supported OpenGL version isn't recent enough for our PBR shader.  @bigbike I assume the PBR shader requires 4.1 (reasonable if it does, just double checking)?  Is there a way to force things to use the Phong shader via a config? (CC @jturner65 I think).

JiayunjieJYJ, 2021-07-09:
> Looks like your supported OpenGL version isn't recent enough for our PBR shader. @bigbike I assume the PBR shader requires 4.1 (reasonable if it does, just double checking)? Is there a way to force things to use the Phong shader via a config? (CC @jturner65 I think).

@erikwijmans 
Thank you for your patience. I dont know the reason about that, or it only occurs on laptop (maybe, I dont know). 
The Ubuntu18.04.5 virtual machine on my laptop is of 4G memory, 4 processors and 40G hard disk space.

Thank you a lot!

JiayunjieJYJ, 2021-07-09:
> Looks like your supported OpenGL version isn't recent enough for our PBR shader. @bigbike I assume the PBR shader requires 4.1 (reasonable if it does, just double checking)? Is there a way to force things to use the Phong shader via a config? (CC @jturner65 I think).

@erikwijmans 
This is my OpenGL verison:
```
OpenGL   GL_VENDOR：VMware, Inc.
GL_RENDERER ：SVGA3D; build: RELEASE;  LLVM;
GL_VERSION：3.3 (Compatibility Profile) Mesa 20.0.8
GLU_VERSION：1.3
```

JiayunjieJYJ, 2021-07-10:
> Looks like your supported OpenGL version isn't recent enough for our PBR shader. @bigbike I assume the PBR shader requires 4.1 (reasonable if it does, just double checking)? Is there a way to force things to use the Phong shader via a config? (CC @jturner65 I think).

@erikwijmans Hello, May I ask a question about how to update OpenGL in my ubuntu virtual machine?  Cause I use apt upgrade and apt update, they are both useless, I wonder whether the version of OpenGL is limited by my laptop, then I cant update OpenGL? 

Thanks!

erikwijmans, 2021-07-11:
I'm not familiar with running VMs so I don't know

JiayunjieJYJ, 2021-07-12:
> I'm not familiar with running VMs so I don't know

Alright, thank you anyway. So I consider change anothor computer to install it, could you please tell my **what's the full requirements of the computer** to install the habitat-lab and habitat-sim? Then I can pick a proper device.

Thanks!

erikwijmans, 2021-07-12:
OpenGL>=4.1 (I haven't heard of anything recent besides VMs not supporting this as OpenGL 4.1 was release over 10 years ago) and the ability to install some standard packages (most systems already have them installed so I don't have the full list as things like a GUI depend on them also).  Headless mode requires an nvidia GPU and nvidia's OpenGL libs.

closed 2021-08-16 by dhruvbatra