facebookresearch/habitat-sim #1378: Empty RigidBodyManager when there are no sensors

Jiayuan-Gu, 2021-07-11:
## Habitat-Sim version
v0.2.0

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

```python
import habitat_sim

sim_cfg = habitat_sim.SimulatorConfiguration()
sim_cfg.enable_physics = True
agent_cfg = habitat_sim.AgentConfiguration()
#camera_sensor_spec = habitat_sim.CameraSensorSpec()
#agent_cfg.sensor_specifications = [camera_sensor_spec]
hab_cfg = habitat_sim.Configuration(sim_cfg, [agent_cfg])
with habitat_sim.Simulator(hab_cfg) as sim:
    obs = sim.get_sensor_observations()
    print('obs', obs.keys())
    sim.step_physics(-1)
    rigid_obj_mgr = sim.get_rigid_object_manager()
    print('rigid_obj_mgr', rigid_obj_mgr)
```

## Expected behavior

`rigid_obj_mgr` is `None` when there are no sensors.

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux): Linux-5.4.0-77-generic-x86_64-with-debian-buster-sid
 - How you installed PyTorch (`conda`, `pip`, source): pip
 - Build command you used (if compiling from source): 
 - Python version: 3.6
 - GPU models and configuration: 1080Ti
 - CUDA version: 11.2
 - PIP/conda dependencies packages versions
 - Any other relevant information:

dhruvbatra, 2021-07-12:
CC: @jturner65

erikwijmans, 2021-07-12:
Can you see if this is fixed on the nightly builds?  #1308 should have fixed this

Jiayuan-Gu, 2021-07-12:
@erikwijmans The nightly build works. Thx. 
BTW, do you know how to disable the logging like `BulletPhysicsManager::addArticulatedObjectFromURDF: simpleObjectHandle :  hab_fetch`, which seems not be able to suppressed by `GLOG_minloglevel=2 MAGNUM_LOG=quiet`?

Jiayuan-Gu, 2021-07-12:
<https://github.com/facebookresearch/habitat-sim/blob/aeaa25ceb7700e71331ee11d6472f79cbd4e671d/src/esp/physics/bullet/BulletPhysicsManager.cpp#L198>
It comes from `Magnum::Debug`

aclegg3, 2021-08-05:
PR #1411 addresses configurable subsystem logging and should be merged soon.

closed 2021-08-05 by aclegg3