facebookresearch/habitat-sim #1379: Construct my own scenes and insert my own objects

phj128, 2021-07-12:
## Habitat-Sim version
v0.2.0

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Hi, I would like to use Habitat-sim to construct my own scene with mesh. Also, I would like to insert some objects and other things like human to the existing scene. Is there any guidance about this?

erikwijmans, 2021-07-12:
Please take a look at https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/managed_rigid_object_tutorial.ipynb and other tutorials https://aihabitat.org/docs/habitat-sim/

phj128, 2021-07-14:
Thanks for your help.

phj128, 2021-07-20:
Hi, I have already generated the navmesh as the tutorial based on my own scene, and run habitat-viewer with my own scene, 

```
habitat-viewer /path/to/my/own/scene.glb
```

At the start, the simulation seems good, but when I move the camera quickly or move the agent to some postions, I encountered the following errors.

Assertion measurement._movingSum >= currentMeasurementData failed at /opt/conda/envs/py36/conda-bld/habitat-sim_1625040415038/work/src/deps/magnum/src/Magnum/DebugTools/FrameProfiler.cpp:199
[1]    18653 abort (core dumped)  habitat-viewer ../path/to/my/own/scene.glb

erikwijmans, 2021-07-20:
Please make a new issue.  This seems unrelated.

closed 2021-07-20 by erikwijmans