facebookresearch/habitat-sim #1383: Sensor readings always deterministic?

ericchen321, 2021-07-14:
## Habitat-Sim version
0.2.0

## Docs and Tutorials
Did you read the docs? Yes

Did you check out the tutorials? Yes

## ❓ Questions and Help

Hello, we are trying to figure out if the Habitat environment has randomness or not. 

So we tried to get sensor readings from an environment by calling ``Env.reset()`` and ``Env.step()`` (we placed an RGBD agent inside to navigate over an episode), and we found that both RGB and depth readings produced by the environment are exactly the same across multiple runs, despite us setting different random seeds with ``Env.seed()``. 

This seems to be contradictory to what has been observed in [this issue](https://github.com/facebookresearch/habitat-sim/issues/689#issue-658097171) from July 2020, where the RGB output from the environment has some random noise. So we thought maybe we accidentally configured something to make the environment stay deterministic?

erikwijmans, 2021-07-14:
It depends on if you have a noise model enabled. If you don't have a noise model enabled the random seed doesn't influence rendering.

ericchen321, 2021-07-14:
Oh okay I see, that makes sense. Thanks Erik!

closed 2021-07-14 by ericchen321