facebookresearch/habitat-sim #1385: ModuleNotFoundError: No module named 'magnum.scenegraph'

emontero1991, 2021-07-14:
Steps to install : 
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim; git checkout 9575dcd45fe6f55d2a44043833af08972a7895a9; 
pip install -r requirements.txt; 
python3 setup.py install 
python3 -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path /path/to/data/

Python3.6.9
Ubuntu 18.04

Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/msdl/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/msdl/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/msdl/habitat-sim/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/msdl/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/msdl/habitat-sim/habitat_sim/agent/agent.py", line 11, in <module>
    import magnum.scenegraph
ModuleNotFoundError: No module named 'magnum.scenegraph'

Actions performed:
pip3 install build/deps/magnum-bindings/src/python/       force reinstall magnum 
Got stuck in loop between magnum 0.0.0 and magnum 12.0.0 with following errors respectively.

ImportError: /usr/local/lib/python3.6/dist-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _Py_ZeroStruct

ModuleNotFoundError: No module named 'magnum.scenegraph'

Please help me to solve the issue 
Thank you so much

emontero1991, 2021-07-14:
Actions performed:
1) pip3 install build/deps/magnum-bindings/src/python/                 To force reinstall magnum

2) python3 setup.py install --skip-installed magnum

3) Tried uninstalling habitat-sim and magnum,  first installed magnum and then installed habitat-sim again

Successfully installed magnum-0.0.0 gives this error

Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/msdl/habitat-sim/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/msdl/habitat-sim/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/msdl/habitat-sim/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/msdl/habitat-sim/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/msdl/habitat-sim/habitat_sim/agent/agent.py", line 10, in <module>
    import magnum as mn
  File "/usr/local/lib/python3.6/dist-packages/magnum/__init__.py", line 28, in <module>
    from _magnum import *
ImportError: /usr/local/lib/python3.6/dist-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _Py_ZeroStruct

erikwijmans, 2021-07-14:
When using setup.py, you'll need to run that command outside the root repo folder, otherwise you'll pick up the source instead of what was installed, possibly causing conflicts.

```
undefined symbol: _Py_ZeroStruct
```
generally indicates that the wrong version of python was used for compiling.  This can be most easily fixed by using some python environment, virtualenv, conda, etc.

emontero1991, 2021-07-20:
Thank you so much for the response.
We got succeeded in the installation of Habitat-sim.

By using a virtual python environment problem got resolved. (venv).

closed 2021-08-16 by dhruvbatra