facebookresearch/habitat-sim #1393: Unable to replicate #1351. Suspect coordinate bug.

dhruvbatra, 2021-07-18:
## Habitat-Sim version
Tested on both v0.2.0 and master. 

## 🐛 Bug

Trying to replicatd what @aclegg3 shows in #1351 video. Unable to click on lamp (nothing happens). 

Accidentally clicking around, I discovered that I was able to open cabinets but by pointing the mouse at a offset (see screenshot). I suspect there is a coordinate transform bug or something wrong with the object coordinates.  

https://user-images.githubusercontent.com/2941091/126056865-962b3ca4-8668-4929-bfb2-f868542664c0.mov

## Steps to Reproduce

Steps to reproduce the behavior:

1. Conda install habitat_sim (on macOS)
2. python -m habitat_sim.utils.datasets_download --uids replica_cad_dataset replica_cad_baked_lighting --data-path /path/to/data/
3. 

## System Info

ENVIRONMENT INFO:
Platform: Darwin-20.5.0-x86_64-i386-64bit
Machine: x86_64
Processor: i386
Libc version:
Mac version: 10.16
Python version: 3.6.13
Architecture: 64bit
Win version:
System OS: Darwin
Release: 20.5.0
Version: Darwin Kernel Version 20.5.0: Sat May  8 05:10:33 PDT 2021; root:xnu-7195.121.3~9/RELEASE_X86_64
Operational System: darwin
GCC version: b'Apple clang version 12.0.5 (clang-1205.0.22.11)\n'
CMAKE version: b'cmake version 3.14.0\n'
NVIDIA-SMI: b''
Pip packages versions:
b'imageio-ffmpeg==0.4.4\nmatplotlib==3.3.4\nnumba==0.53.1\nnumpy-quaternion==2021.6.9.13.34.11\ntqdm==4.61.2\n'
Conda packages versions:
b'attrs                     21.2.0             pyhd3eb1b0_0    19.1.0\ngitpython                 3.1.18             pyhd3eb1b0_1  \nimageio                   2.9.0                    pypi_0    pypi\nimageio-ffmpeg            0.4.4                    pypi_0    pypi\nmatplotlib                3.3.4                    pypi_0    pypi\nnumba                     0.53.1                   pypi_0    pypi\nnumpy                     1.19.5                   pypi_0    pypi\nnumpy-quaternion          2021.6.9.13.34.11          pypi_0    pypi\npillow                    8.3.1                    pypi_0    pypi\nscipy                     1.5.4                    pypi_0    pypi  1.3.0\ntqdm                      4.61.2             pyhd3eb1b0_1  \n'

jturner65, 2021-07-20:
Seems like you have the old dataset. This is my fault, I was unaware I needed to update the download script for it to get the most recent dataset. Doing so now.

jturner65, 2021-07-28:
I am unable to reproduce the problem with the current 1.1 dataset on both Ubuntu and Mac (Big Sur 11.4), [as shown in this thread on Slack](https://cvmlp.slack.com/archives/C01QHBYJH7X/p1627409574132100).

jturner65, 2021-07-29:
This issue is apparently caused by DPI scaling not being properly accounted for in the ray casting in viewer.

jturner65, 2021-07-29:
Addressed in this PR : https://github.com/facebookresearch/habitat-sim/pull/1415

aclegg3, 2021-08-05:
Closing this with the above solution. Feel free to re-open if you are still experiencing issues after the patch.

closed 2021-08-05 by aclegg3

dhruvbatra, 2021-08-05:
Yup, this is resolved by #1415.