facebookresearch/habitat-sim #1396: No warning when default.physics_config.json is missing

Jiayuan-Gu, 2021-07-20:
## Habitat-Sim version
master

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

1. `conda install habitat-sim withbullet headless -c conda-forge -c aihabitat-nightly`
2. `python -m habitat_sim.utils.datasets_download --uids ci_test_assets --data-path data`
3.  Run the following 
```python
import habitat_sim

def test_physics_config_file(physics_config_file):
    sim_cfg = habitat_sim.SimulatorConfiguration()
    sim_cfg.enable_physics = True
    if physics_config_file:
        sim_cfg.physics_config_file = physics_config_file
    agent_cfg = habitat_sim.AgentConfiguration()
    camera_sensor_spec = habitat_sim.CameraSensorSpec()
    camera_sensor_spec.uuid = 'rgb'
    camera_sensor_spec.sensor_type = habitat_sim.SensorType.COLOR
    camera_sensor_spec.resolution = [480, 640]
    camera_sensor_spec.position = [1.5, 0.5, 0.0]
    camera_sensor_spec.orientation = [-0.1, 1.57, 0.0]

    agent_cfg.sensor_specifications = [camera_sensor_spec]
    hab_cfg = habitat_sim.Configuration(sim_cfg, [agent_cfg])
    with habitat_sim.Simulator(hab_cfg) as sim:
        art_obj_mgr = sim.get_articulated_object_manager()
        urdf_path = "data/robots/hab_fetch/robots/hab_fetch.urdf"
        robot = art_obj_mgr.add_articulated_object_from_urdf(urdf_path, fixed_base=True)

test_physics_config_file('wrong/path/to/physics_config_file')
```

## Expected behavior

```
addArticulatedObjectFromURDF not implemented in base PhysicsManager.
E0720 17:37:50.838002  3802 ManagedContainerBase.h:203] ::getObjectHandleByID : Unknown ArticulatedObject managed object ID:-1. Aborting
E0720 17:37:50.838011  3802 ManagedContainerBase.h:331] <ArticulatedObject>::getObjectCopyByID : Unknown ArticulatedObject managed object handle :. Abortin
```
It seems that the physic backend is not set correctly due to the missing config file.
Two suggestions to improve:
1. Some warnings should be raised to inform the users of the missing config file, otherwise it is quite subtle.
2. How to download default physics_config_file can be highlighted somewhere, or automatically downloading it should be enabled.

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux): Linux-5.4.0-77-generic-x86_64-with-debian-buster-sid
 - How you installed PyTorch (`conda`, `pip`, source): pip
 - Build command you used (if compiling from source): 
 - Python version: 3.6
 - GPU models and configuration: 1080Ti
 - CUDA version: 11.2
 - PIP/conda dependencies packages versions
 - Any other relevant information:

dhruvbatra, 2021-07-20:
CC: @jturner65

aclegg3, 2021-08-02:
Note that the default behavior with a missing config is improved by #1409.
We should still add a better warning for users expecting to have their own file picked up.