facebookresearch/habitat-sim #1401: AgentState.linear_velocity and AgentState.angular_velocity return 0-vectors

ericchen321, 2021-07-22:
## Habitat-Sim version
v0.2.0

## Docs and Tutorials
Did you read the docs? Yes

Did you check out the tutorials? Yes

## ❓ Questions and Help

### What I was trying to do
Hello, while playing around with the rigid object tutorial, I was trying to access the embodied agent's linear and angular velocity after setting them to some non-zero values. Specifically I have the following code:
```
# [embodied_agent]
# load the lobot_merged asset
locobot_template_id = obj_templates_mgr.load_configs(
    str(os.path.join(data_path, "objects/locobot_merged"))
)[0]
...
vel_control = sim.get_object_velocity_control(id_1)
vel_control.linear_velocity = np.array([0, 0, -1.0])
vel_control.angular_velocity = np.array([0.0, 2.0, 0])
...
vel_control.controlling_lin_vel = True
vel_control.controlling_ang_vel = True
vel_control.lin_vel_is_local = True
vel_control.ang_vel_is_local = True

# simulate forward and turn
observations += simulate(sim, dt=1.0, get_frames=make_video)
print(f"linear velocity: {sim.agents[0].get_state().velocity}")
print(f"angular velocity: {sim.agents[0].get_state().angular_velocity}")
```
Some of the API calls in the snippet are from Sim V1 (`Simulator.get_object_velocity_control()` for example), but with the V2 API calls I had the same problem.
### Expected behavior
I thought since  `simulate()` does not modify the agent's velocity, the print statements should show the velocities as the exact same values as the ones I set. 
### Actual behavior
I got the following values printed out:
```
linear velocity: [0. 0. 0.]
angular velocity: [0. 0. 0.]
```
So I am wondering why the velocities are both zero. Thanks in advance for the help.

Skylion007, 2021-07-22:
This is a good catch, those two properties are from an earlier API and are currently unused. We need to either add a pass through to the underlying physics velocity or remove those and calculate the velocity from the attached object to the agent. https://github.com/facebookresearch/habitat-sim/blob/55cf55827e21aa6816332ab3cae5b644b348f0a0/habitat_sim/agent/agent.py#L84 We are also working on merging the duplicated Python and C++ classes to reduce this code duplication. In other words, this info is not available in the agent_state class, but it should be.  

TLDR: Don't use that velocity property on agent_state, we are not updating that value, but we should be.
@aclegg3 @eundersander flagging this bug.

ericchen321, 2021-07-22:
Thanks for the help, Aaron and Alex! Looking forward to the update.

closed 2022-02-03 by aclegg3