facebookresearch/habitat-sim #1418: Bird eye view/ Top view

aveen-d, 2021-08-03:
Hi, do you have a bird eye view depth estimate camera, just like the ones used for autonomous vehicles? 

[image]
something like this, images from multiples cameras are combined to form bird eye image.

dhruvbatra, 2021-08-03:
CC: @bigbike

aclegg3, 2021-08-05:
I'm sure it is possible to construct a top-down view from combined camera input, but it sounds complicated to create the image you show here (e.g. some occluded areas such as the space between the houses would not be known from agent camera images). You are of course welcome to contribute this functionality and we would be happy to answer questions and assist with code review and such.

An alternative approach may be to add an orthographic sensor above the agent pointing down and render an image like this. Note that you may get clipping from the ceiling and such for indoor scenes.

At this time the closest thing we currently have to this kind of map is a utility for generating a grid style top down map of the navigation mesh as explained in [this tutorial segment](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb#scrollTo=5puIHoiuFkhV). You are also welcome to enhance this utility to fit your needs.

closed 2021-08-16 by dhruvbatra