facebookresearch/habitat-sim #1424: What's the original size of the template object (e.g. cylinder) before scaling?

AgentEXPL, 2021-08-05:
I know that the size of the object can be changed by changing the template scale value. To setting an accurate object size, I need to know the original size of the template for computing the accurate scale value (i.e., [x,y,z] as follows). Is there any explanations/tutorials providing the original size of different template objects?

---------------------------------------------------------------------------------------------------------------
cylinder_template_handle = obj_attr_mgr.get_template_handles("cylinderSolid")[0]
obj_template_handle = cylinder_template_handle
obj_template = obj_attr_mgr.get_template_by_handle(obj_template_handle)
obj_template.scale = [x, y, z]

aclegg3, 2021-08-05:
Great question, at this time there is no recorded meta-data for the initial size of any object. I'd recommend a pre-process on the objects you are interested in to compute and save that information as metadata. 

You could do this at runtime by first adding the object, querying its size, removing, scaling, and finally re-adding it. This should not take much time compared to loading in the asset initially. You could also cache a local dictionary with the info once you compute it the first time.

You can also use the `"user_defined"` tag to add custom fields to the object attributes JSON configs if you want to pass these into Habitat-sim for later query after computing them the first time. See [this JSON config test](https://github.com/facebookresearch/habitat-sim/blob/8669883a6d517d8daba9458e1b40f5a19b71f0cc/src/tests/AttributesManagersTest.cpp#L951) for an example of adding user defined values to an object config. These can be queried at runtime from the object's `user_attributes` property.

AgentEXPL, 2021-08-06:
Thanks for your good suggestion. One more question. Is it correct that the object size is computed from the "collision_shape_aabb" at runtime?
[image]

aclegg3, 2021-08-06:
> Is it correct that the object size is computed from the "collision_shape_aabb" at runtime?

The bounding box stored in the object's SceneNode (object.[root_scene_node](https://aihabitat.org/docs/habitat-sim/habitat_sim.physics.ManagedRigidObject.html#root_scene_node).[cumulative_bb](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneNode.html#cumulative_bb) is computed from the render shape geometry at runtime by aggregating the [mesh_bb](https://aihabitat.org/docs/habitat-sim/habitat_sim.scene.SceneNode.html#mesh_bb) for each component of the asset's SceneGraph subtree (e.g., each separate mesh in the glTF hierarchy). The each component's mesh_bb is computed from its render shape geometry in local space.

The `collision_shape_aabb` is computed by Bullet and includes the aggregate bounding box of all collision geometry components as well as the collision margin.

It depends on your use case which of these would be best for you but both are available when Bullet is enabled immediately after an object has been added to the scene.

AgentEXPL, 2021-08-09:
Thanks for the clear explanation. I have no more questions on this topic.

closed 2021-08-09 by AgentEXPL