facebookresearch/habitat-sim #1445: How to place objects directly on the ground without clipping through the stage?

AditMeh, 2021-08-21:
## Habitat-Sim version

I am running https://github.com/facebookresearch/habitat-sim/blob/master/examples/colab_utils/colab_install.sh in a colab notebook using the following command:

`!curl -L https://raw.githubusercontent.com/facebookresearch/habitat-sim/master/examples/colab_utils/colab_install.sh | NIGHTLY=true bash -s`

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Yes

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Yes, a lot of my code was taken from the tutorials

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

I have a colab notebook with a minimum reproducible example, you may need to run the final block twice (it crashes sometimes). Note that a lot of code was taken from the tutorials. 

https://colab.research.google.com/drive/1mb5ist-Y2VVUX-uFcX2RBQkc2pCYo3jo?usp=sharing

To summarize, I am first setting the stage floor test asset from habitat-sim as my stage for the scene and creating a standard camera agent. After this, I place the agent and the nested_box test asset from habitat-sim at the origin. Then I move the agent -2.0 in the x-direction and have it look at the box using the `move_agent_to_coordinates` function, ending off by displaying what the camera is looking at. 

As you can see in the image, the nested box is clipping through the stage. Additionally, the printed bounding box has one of its points with a negative y value - a corner of the box has been clipped below the stage.

How can I get the box to precisely be placed on top of the stage as in the real world? I don't know if this is an issue with the stage (the stage may be pushed up or not perfectly flat) or with the way I placed the box.

aclegg3, 2021-08-21:
Hey, @AditMeh. 

Sounds like what you'd like to do is place objects in the scene at physically valid locations (like the floor or counter tops) without clipping. Take a look at our [ECCV 2020 Tutorial series  ](https://aihabitat.org/tutorial/2020/), specifically the **Habitat-Sim for Interaction** section which demonstrates how to place objects from the navmesh using rejection sampling. There is example code runnable in Colab to get you started.

AditMeh, 2021-08-21:
I checked out the sample_object_state function in the notebook you linked https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Interactivity.ipynb#scrollTo=FbZbSMVpXKFJ&line=60&uniqifier=1, however, this seems to be for random points. How could I adapt this to "snap" an object to the stage mesh given a point in the world coordinate frame?

erikwijmans, 2021-08-22:
`sim.pathfinder.snap_point` is likely what you are looking for.

closed 2022-03-01 by aclegg3