facebookresearch/habitat-sim #1450: Unable to see lighting changes

chengine, 2021-08-24:
Hello, 

I am trying to run the Lighting Setup tutorial in `examples/tutorials/lighting_tutorial.py` . However, the images show no lighting changes, unlike what is shown in the documentation page Working with Lights. This is run off a freshly installed conda environment. How can I fix this?

Thanks!

aclegg3, 2021-08-24:
Hey @chengine I just ran the tutorial code on my machine. You are correct that the results are not matching the older version. We have made several changes to the system since this page was published and we should update that. 

However, I do see lighting changes demonstrating that configurable lighting is working. For example `8.jpg` (below) with green light on one chair and another unlit. Are you seeing these results?
[image]

chengine, 2021-08-25:
Hello. Thanks so much for the reply. Yes, I can replicate the configurable lighting. Is there a way to change lighting settings in the new system? More specifically for my needs, is there a way in the new system to activate global lighting (i.e. `cfg.sim_cfg.scene_light_setup = habitat_sim.gfx.DEFAULT_LIGHTING_KEY`)? Thank you.

ybbbbt, 2021-09-06:
I tried to add `cfg.sim_cfg.override_scene_light_defaults = True` when setting the `cfg.sim_cfg.scene_light_setup`, which should be able to change the scene global lighting.

xinranliang, 2022-05-23:
Hi! I tried to follow the updated tutorial [here](https://github.com/facebookresearch/habitat-sim/pull/1760/commits/4308327a000ef23e8b95bdbe0a6ffce18ee54765) by setting `cfg.sim_cfg.override_scene_light_defaults = True`. But it reports error `AttributeError: 'habitat_sim._ext.habitat_sim_bindings.SimulatorConfiguration' object has no attribute 'override_scene_light_defaults'`. I'm still unable to overwrite default lighting setup.

I'd like to set up custom lighting in the global environment. Is there a way that I can either first set up as default key then overwrite, or directly set up a custom light info? Since current tutorial only talks about first setting up default key and then change it by reconfigure sim settings.

For reference, I'm using a previous version of `habitat-sim` for the purpose of getting semantic map: branch at `tags/v0.1.5`. I'm not sure whether this will make a difference. Thanks so much for help!

aclegg3, 2022-05-23:
@xinranliang that's a pretty old version of Habitat, so `override_scene_light_defaults` is not available or necessary. You should checkout [the change-log for that version](https://github.com/facebookresearch/habitat-sim/releases/tag/v0.1.5) and look into the PRs regarding Phong Shading and configurable lighting to get a feel for how to manipulate the light setups in that version.

closed 2022-08-31 by aclegg3