facebookresearch/habitat-sim #1463: Agent gets stuck in scene when climbing upstairs or downstairs

quintus0505, 2021-09-01:
Hi 

I have successfully build .navnesh file from .glb (From HM3D), but when I use habitat-viewer to checkout the scene, in some cases the agent can get stuck when going upstairs or downstairs. Is  there any solutions?

aclegg3, 2021-09-01:
In some scans from HM3D, environmental geometry (low ceilings, narrow passages, or small holes in the ground) can prevent the default agent (1.5m tall) from fitting through enclosed spaces. We saw this on several stairways from HM3D. We tried to screen any issues that were not fixed by shrinking the agent slightly. 

Try recomputing the NavMesh with a shorter agent (e.g, 1.3 meters). You can learn a bit about the NavMesh and configuring its parameters in our [Habitat-sim Basics for Navigation](https://colab.research.google.com/github/facebookresearch/habitat-sim/blob/master/examples/tutorials/colabs/ECCV_2020_Navigation.ipynb) tutorial. There is a Colab cell which allows you to experiment with recomputing the NavMesh interactively. While this tutorial is written in python, the C++ interface is very similar if you'd like to modify the viewer to customize the navmesh. Alternatively you could pre-compute a suitable NavMesh in python and save it, then load it in the viewer using a console argument (`--navmesh-file`). Run the viewer application with no arguments for more information.

quintus0505, 2021-09-02:
Thanks for quick response!
I tried to run codes like those in Habitat-sim Basics for Navigation tutorial (Recompute NavMesh part) but it seems that I cannot save any change(I always get agent_height = 1.5 when loading from navmesh)

then I managed to find the default settings in ```habitat-sim/src/esp/nav/PathFinder.h```. After changing the agentHeight to 1.3f I rebuild the datatool and rebuild all the .navmesh. But still it makes no difference, I wonder whether I missed some important steps when creating navmesh by using datatool or recomputing navmesh?

Thanks

aclegg3, 2022-03-01:
@quintus0505 I suggest making your modifications directly to the [NavMeshSettings](https://aihabitat.org/docs/habitat-sim/habitat_sim.nav.NavMeshSettings.html) and recomputing in python dynamically instead of the datatool. So, instead of generating the navmeshes, saving, and re-loading, you would re-generate them during runtime while you debug the issue. You may want to try using the python [viewer.py ](https://github.com/facebookresearch/habitat-sim/blob/main/examples/viewer.py)application so you can visualize the results and test the stairs during your work.

closed 2022-03-01 by aclegg3