facebookresearch/habitat-sim #1464: Habitat-sim bug on gibson semantics

gianscarpe, 2021-09-02:
## Habitat-Sim version
0.2.1

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? The bug you are about to report may already be fixed in the latest version.

Master branch contains 'bleeding edge' code, but we do appreciate bug reports for it!

## 🐛 Bug
I found a bug in the semantic annotation of at least one scene from gibson. I'm referring to "Swarmville"
This is an example of the bug:
What you see here is the juxtaposition of semantic, labels, and Rgb data for swarmville scene
[image]

Here's from another point of view. RGB, depth, and semantic
[image]
[image]
[image]

## Steps to Reproduce

Steps to reproduce the behavior:

1. Get the semantic annotation from https://3dscenegraph.stanford.edu/Swormville.html using the provided datatool 
2. Collect random samples fr
3. Generate random samples from the dataset

 Please note that without a minimal working example to reproduce the bug, we may not be able to help you.

## Expected behavior

## Additional context

## System Info

Please copy and paste the output from the
[environment collection script](https://github.com/facebookresearch/habitat-sim/tree/master/habitat_sim/utils/collect_env.py)
(or fill out the checklist below manually).

You can run the script with:
```
# For security purposes, please check the contents of collect_env.py before running it.
python habitat_sim/utils/collect_env.py
```

 - OS (e.g., Linux):
 - How you installed PyTorch (`conda`, `pip`, source):
 - Build command you used (if compiling from source):
 - Python version:
 - GPU models and configuration:
 - CUDA version:
 - PIP/conda dependencies packages versions
 - Any other relevant information:

erikwijmans, 2021-09-02:
This seems to be an issue with the Gibson semantic annotations themselves. It seems like semantics were labelled on a different mesh and this used for RGB, so that phantom is present in the semantic mesh but not the RGB mesh.

gianscarpe, 2021-09-05:
Ehi @erikwijmans , thanks for your reply! Do you know of any other ways to check this inconsistency?

closed 2022-04-21 by aclegg3