facebookresearch/habitat-sim #1467: I have a problem. I tried to add camera configuration to agent configuration in the following code:

Gricha1, 2021-09-07:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Master branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Skylion007, 2021-09-09:
If the colab kernal dies, it's proibably segfault. The most common reason is an invalid config / environment (like encountering a filenotfound error when loading the scene, invalid configuration, etc...)

Gricha1, 2021-09-20:
On the example of using habitat sim on google colab the authors are using google colab's GPU. So the problem was that i didnt use gpu in colab, thanks :)

closed 2021-09-20 by Gricha1