facebookresearch/habitat-sim #1485: Region segmentation sensor

zhangfuyang, 2021-09-16:
Hi,

I need a new sensor that gets instance **region** segmentation, meaning that labeling each image pixel with instance room type. I guess it's feasible since we have region annotation. The problem is if I project each pixel to the global coordinate and evaluate it if it's inside the region by python is very slow.

I wonder if there is a better way to do it. Maybe a brief explanation about how "SEMANTIC" sensor works would help. I can mimic the code for semantic sensor. I guess they are very similar. 

Thank you!

erikwijmans, 2021-09-16:
The way the semantic sensor works is with a special mesh with instance annotations. 

A faster way than projecting into 3D is to create a mapping that goes Instance ID to Region ID. Then you can do `region_seg = inst_to_region[inst_seg]`, which may be fast enough. Note that this mapping will be specific to each scene.

zhangfuyang, 2021-09-16:
Oh Right!! That should be the fastest way. Thank you Erik

closed 2021-09-16 by zhangfuyang