facebookresearch/habitat-sim #1491: Accessing Mesh of Articulated URDF Object

ugurbolat, 2021-09-19:
## Habitat-Sim version
vx.x.x

Habitat is under active development, and we advise users to restrict themselves to stable releases. Are you using the latest [release version](https://github.com/facebookresearch/habitat-sim/releases) of Habitat-Sim? Your question may already be addressed in the latest version. We may also not be able to help with problems in earlier versions because they sometimes lack the more verbose logging needed for debugging.

Main branch contains 'bleeding edge' code and should be used at your own risk.

## Docs and Tutorials
Did you read the docs? https://aihabitat.org/docs/habitat-sim/

Did you check out the tutorials? https://aihabitat.org/tutorial/2020/

Perhaps your question is answered there. If not, carry on!

## ❓ Questions and Help

Is it possible to access mesh of an ManagedArticulatedObject that I load with add_articulated_object_from_urdf() ?

aclegg3, 2021-09-20:
It is not currently possible to access render or collision mesh data from the python API.

ugurbolat, 2021-09-20:
@aclegg3 thanks for the reply!

I've just also seen the related issue - #1081 and PR - #1149.

Closing as it is a duplicate.

closed 2021-09-20 by ugurbolat