facebookresearch/habitat-sim #1497: Problems with GreedyGeodesicFollower

weders, 2021-09-22:
## Habitat-Sim version
v0.2.1

I am currently working on creating trajectories in Habitat AI. I do that using the following steps:

1. Sample key frame positions using:
`    key_frame_positions = [sim.pathfinder.get_random_navigable_point() for kf in range(args.n_key_frames)]
`
2. Sorting the sampled key frames according to the geodesic distance obtained from shortest path.
3. Creating a list of actions that connects these key frames using GreedyGeodesicFollower.

Unfortunately, this approach frequently leads to GreedyGeodesicError in step 3. However, I do not understand how this can happen. Step 1 and 2 make sure that the sampled start and end points are navigable and that there exists a shortest point between the two points. Otherwise, the points get already filtered in these steps. 

I also tried to catch the error and adjust the point by snapping it to the closest navigable point. I.e.:

```
try:
    actions += follower.find_path(end_point)
except:
    end_point = sim.pathfinder.snap_point(end_point)
    actions += follower.find_path(end_point)
```

However, snap_point always returns the same point. How can this be?

Thank you very much!

erikwijmans, 2021-09-23:
What action space and agent size are you using? With the default agent size and action space, the follower isn't perfect but gets there around 99% of the time in my testing.

> Unfortunately, this approach frequently leads to GreedyGeodesicError in step 3. However, I do not understand how this can happen.

This can happen as the agent's action space may not allow it to follow the geodesic shortest path -- the geodesic shortest path assumes an agent with an omnidirectional drive. Unfortunately computing the shortest path in the agents action space is extremely computationally expensive.

> However, snap_point always returns the same point. How can this be?

snap_point will return the same point if the input point is already navigable. Since you sampled the point from the navmesh, it's already navigable.

weders, 2021-09-24:
Thanks a lot for the answers, I think I understand the problem better now!

closed 2021-09-24 by weders

tekinengin, 2022-03-10:
I have a similar issue where i find the shortest path to a location using habitat_sim.ShortestPath but then ShortestPathFollower gives _stop_on_error while getting next action. 

Agent action space contains forward, turn_right and turn_left. I tried with very fine actions (FORWARD_STEP_SIZE:0.05 and TURN_ANGLE:1) but it still behaves same. 

In below example i spawn the agent to a certain location to reproduce error. 

Habitat-sim : 0.1.6
Gibson Scene: Sherrill

What would be the way to solve this issue?

[image]