facebookresearch/habitat-sim #1500: Ground Truth semantic annotation of Gibson?

zenght, 2021-09-27:
Hi, I want to know how to get semantic annotations corresponding to Gibson datasets. I followed the method to obtain the corresponding images [https://aihabitat.org/docs/habitat-api/habitat-api-demo.html#scene-semantic-annotations](url). but the image seems to exist only two values. 
[image]

[image]
And the dataset I used is Habitat -Gibson dataset.

dhruvbatra, 2021-09-27:
Those likely are the only annotations available in the Gibson dataset. Try MP3D.

closed 2021-09-27 by dhruvbatra

vincent341, 2022-02-24:
> Hi. I'm struggling with creating 2D semantic maps for MP3D. Would you mind letting me know how to get 2D top-down ground truth of semantic maps of MP3D? Thanks very much.

vincentcartillier, 2022-02-24:
Hi @vincent341, (duplicate from [here](https://github.com/facebookresearch/habitat-sim/issues/948)) 
You can check the following project: [Semantic-MapNet](https://github.com/vincentcartillier/Semantic-MapNet) in which we built such top-down semantic maps for MP3D.

Specifically you should check the following scripts:
```
data/build_point_cloud_from_mesh_h5.py
compute_GT_topdown_semantic_maps/build_semmap_from_obj_point_cloud.py
```

vincent341, 2022-02-25:
> Hi @vincent341, (duplicate from [here](https://github.com/facebookresearch/habitat-sim/issues/948)) You can check the following project: [...]

Appreciate your help very much, Vince.