facebookresearch/habitat-sim #1502: Load Navmesh PyBind missing an argument

naokiyokoyama, 2021-09-28:
https://github.com/facebookresearch/habitat-sim/blob/b9df9a6218dd5807e433c3ea8fb6217ca296c420/src/esp/bindings/ShortestPathBindings.cpp#L104

This needs to be able to receive an argument (path to navmesh file).

naokiyokoyama, 2021-09-28:
Made a PR for this [https://github.com/facebookresearch/habitat-sim/pull/1503/commits/b55e4d5c3b1b2760b9238d3754b6598ce42054e8](url)

erikwijmans, 2021-09-28:
It already can accept an argument. pybind automatically infers the arguments. Adding an argument name like you did in #1503 is nice, but it's only necessary for keyword arguments, not positional arguments, i.e.
```
load_nav_mesh(file_path)
```
works on main currently but
```
load_nav_mesh(path=file_path)
```
does not.

closed 2021-09-29 by Skylion007