facebookresearch/habitat-sim #1511: Platform::WindowlessEglApplication::tryCreateContext(): cannot get default EGL display: EGL_BAD_PARAMETER

XUANTONG1999, 2021-10-03:
I tried to render data from Replica dataset with `python examples/example.py --scene`
I have installed the Habitat with

```
conda create -n habitat python=3.6 cmake=3.14.0
conda activate habitat
```

Then I got the error:

```
Platform::WindowlessEglApplication::tryCreateContext(): cannot get default EGL display: EGL_BAD_PARAMETER
WindowlessContext: Unable to create windowless context
```

Could you please help me to solve this issue?

erikwijmans, 2021-10-03:
Please run again with `export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON`

XUANTONG1999, 2021-10-03:
> Please run again with `export MAGNUM_LOG=verbose MAGNUM_GPU_VALIDATION=ON`

I have tried this. Unfortunately, it seems not to work and the same bugs still exist.

erikwijmans, 2021-10-03:
Is there not additional information printed? If so that likely indicates a driver bug or similar.

XUANTONG1999, 2021-10-03:
> Is there no additional information printed? If so that likely indicates a driver bug or similar.

I tried
```
python examples/example.py --scene /Replica-Dataset/dataset/apartment_0/habitat/mesh_semantic.ply --save_png --semantic_sensor --max_frame 20
```

Here is the information printed:
```
sim_cfg.physics_config_file = ./data/default.physics_config.json
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1003 13:42:57.914039 29584 ManagedFileBasedContainer.h:210] <Dataset>::convertFilenameToPassedExt : Filename : default changed to proposed scene_dataset_config.json filename : default.scene_dataset_config.json
I1003 13:42:57.914077 29584 AttributesManagerBase.h:365] <Dataset>::createFromJsonOrDefaultInternal : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I1003 13:42:57.914203 29584 AssetAttributesManager.cpp:120] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1003 13:42:57.914281 29584 AssetAttributesManager.cpp:120] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1003 13:42:57.914352 29584 AssetAttributesManager.cpp:120] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1003 13:42:57.914392 29584 AssetAttributesManager.cpp:120] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1003 13:42:57.914415 29584 AssetAttributesManager.cpp:120] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1003 13:42:57.914434 29584 AssetAttributesManager.cpp:120] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1003 13:42:57.914489 29584 AssetAttributesManager.cpp:120] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1003 13:42:57.914538 29584 AssetAttributesManager.cpp:120] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1003 13:42:57.914566 29584 AssetAttributesManager.cpp:120] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1003 13:42:57.914589 29584 AssetAttributesManager.cpp:120] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1003 13:42:57.914623 29584 AssetAttributesManager.cpp:120] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1003 13:42:57.914656 29584 AssetAttributesManager.cpp:120] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1003 13:42:57.914671 29584 AssetAttributesManager.cpp:108] ::constructor : Built default primitive asset templates : 12
I1003 13:42:57.915589 29584 SceneDatasetAttributesManager.cpp:36] File (default) not found, so new default dataset attributes created and registered.
I1003 13:42:57.915603 29584 MetadataMediator.cpp:127] ::createSceneDataset : Dataset default successfully created.
I1003 13:42:57.915621 29584 AttributesManagerBase.h:365] <Physics Manager>::createFromJsonOrDefaultInternal : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I1003 13:42:57.916173 29584 PhysicsAttributesManager.cpp:26] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I1003 13:42:57.916191 29584 MetadataMediator.cpp:212] ::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
I1003 13:42:57.916199 29584 MetadataMediator.cpp:183] ::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
I1003 13:42:57.916211 29584 MetadataMediator.cpp:68] ::setSimulatorConfiguration : Set new simulator config for scene/stage : /Replica-Dataset/dataset/apartment_0/habitat/mesh_semantic.ply and dataset : default which is currently active dataset.
Platform::WindowlessEglApplication::tryCreateContext(): cannot get default EGL display: EGL_BAD_PARAMETER
WindowlessContext: Unable to create windowless context
```

Sorry I am not sure if this if of a driver bug or not.

erikwijmans, 2021-10-03:
Likely a driver issue. Headless mode requires a recent(ish) nvidia gpu with nvidia's opengl libraries (those are installed by default with a default driver install).

XUANTONG1999, 2021-10-03:
> Likely a driver issue. Headless mode requires a recent(ish) nvidia gpu with nvidia's opengl libraries (those are installed by default with a default driver install).

I see. Would you please tell me if there is anything I can do to handle the driver issue? Do I need to install the OpenGL lib manually?

erikwijmans, 2021-10-03:
You shouldn't need to manually install them, they are installed as part of a normal driver install

XUANTONG1999, 2021-10-03:
> You shouldn't need to manually install them, they are installed as part of a normal driver install

I’m still confused because there are nvidia gpus on my cluster. I suppose the libraries are installed by default. But the driver issue still exists.

erikwijmans, 2021-10-03:
Possibly not. Others have opened issues here because their cluster admins did not install nvidia's opengl libs. Please talk to them.

XUANTONG1999, 2021-10-03:
> Possibly not. Others have opened issues here because their cluster admins did not install nvidia's opengl libs. Please talk to them.

Sure. I really appreciate for your help!

aclegg3, 2022-08-31:
This issue seems to be resolved.

closed 2022-08-31 by aclegg3

jeasinema, 2023-05-06:
Just in case someone needs this: here is a solution for those who don't want to use docker and happen to use a server that does not install `EGL*` along with NVIDIA driver by default.

1. run `nvidia-smi` to check the version of your driver
2. download the driver installation binary from NVIDIA directly. You can do this by searching `nvidia driver <version-id>` on google. The file should be `NVIDIA-Linux-x86_64-<version-id>.run`
3. extract the binary without running. This will create a folder named `NVIDIA-Linux-x86_64-<version-id>`
```bash
sh NVIDIA-Linux-x86_64-<version-id>.run --extract-only   
```
4. create some simlinks
```bash
cd NVIDIA-Linux-x86_64-<version-id>
ln -s ./libEGL.so.<version-id>  libEGL.so.1
```
5. Add this folder to your LD search path
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
```
6. Check your habitat linker (Thanks @erikwijmans for [this](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)!)
```bash
cd
ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")
```
Note that `libEGL.so.1`, `libOpenGL.so.0` and `libGLdispatch.so.0` should be all pointed to the ones in `/path/to/NVIDIA-Linux-x86_64-<version-id>`.

7. You're good to go! Just follow [this](/path/to/NVIDIA-Linux-x86_64-<version-id>) to verify your `habitat-sim` installation.

dhruvbatra, 2023-05-07:
Thanks @jeasinema for this. Would you mind sending a PR to add this information to https://github.com/facebookresearch/habitat-sim/blob/main/BUILD_FROM_SOURCE.md#common-build-issues

donggehan, 2023-09-12:
@jeasinema  Hi, thanks for the information! I'm trying to manually install the missing EGL libraries as you mentioned above. I have driver version 515.105.01 installed (without opengl support), so I downloaded and extracted. 
I see in the NVIDIA-Linux-x86_64-515.105.01 folder that there is libEGL_nvidia.so.515.105.01, do I need to symlink this too? 
(I see in some other threads that libEGL_nvidia.so is necessary so I'm curious if that should be used)
Thanks a lot!

goodstudent9, 2024-02-19:
> Just in case someone needs this: here is a solution for those who don't want to use docker and happen to use a server that does not install `EGL*` along with NVIDIA driver by default. (cross-posted to #851 as well!)
> 
> 1. run `nvidia-smi` to check the version of your driver
> 2. download the driver installation binary from NVIDIA directly. You can do this by searching `nvidia driver <version-id>` on google. The file should be `NVIDIA-Linux-x86_64-<version-id>.run`
> 3. extract the binary without running. This will create a folder named `NVIDIA-Linux-x86_64-<version-id>`
> 
> ```shell
> sh NVIDIA-Linux-x86_64-<version-id>.run --extract-only   
> ```
> 
> 4. create some simlinks
> 
> ```shell
> cd NVIDIA-Linux-x86_64-<version-id>
> ln -s ./libEGL.so.<version-id>  libEGL.so.1 # note: there might be a libEGL_nvidia*, do not symlink it as libEGL.so.1 instead!
> ln -s ./libGL.so.<version> libGL.so.1      
> ```
> 
> 5. Add this folder to your LD search path
> 
> ```
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
> ```
> 
> 6. Check your habitat linker (Thanks @erikwijmans for [this](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)!)
> 
> ```shell
> cd
> ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")
> ```
> 
> Note that `libEGL.so.1`, `libOpenGL.so.0` and `libGLdispatch.so.0` should be all pointed to the ones in `/path/to/NVIDIA-Linux-x86_64-<version-id>`.
> 
> 7. You're good to go! Just follow [this](/path/to/NVIDIA-Linux-x86_64-%3Cversion-id%3E) to verify your `habitat-sim` installation.

That seems like I still can not run well. Following is my wrong message. That seems like Ican not use opencv now.
Traceback (most recent call last):
  File "/data5/xrh/navigation/habitat3/habitat-lab/examples/example.py", line 9, in <module>
    import habitat.gym  # noqa: F401
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/__init__.py", line 9, in <module>
    from habitat.core.benchmark import Benchmark
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/benchmark.py", line 20, in <module>
    from habitat.core.env import Env
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/env.py", line 32, in <module>
    from habitat.tasks.registration import make_task
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/registration.py", line 28, in <module>
    _try_register_rearrange_task()
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/__init__.py", line 12, in _try_register_rearrange_task
    import habitat.tasks.rearrange.actions.actions
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/actions.py", line 17, in <module>
    from habitat.tasks.rearrange.actions.articulated_agent_action import (
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/articulated_agent_action.py", line 2, in <module>
    from habitat.tasks.rearrange.rearrange_sim import RearrangeSim
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 35, in <module>
    from habitat.datasets.rearrange.rearrange_dataset import RearrangeEpisode
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 18, in <module>
    from habitat.datasets.pointnav.pointnav_dataset import PointNavDatasetV1
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 16, in <module>
    from habitat.tasks.nav.nav import (
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/nav/nav.py", line 42, in <module>
    from habitat.utils.visualizations import fog_of_war, maps
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/__init__.py", line 7, in <module>
    from habitat.utils.visualizations import maps, utils
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/maps.py", line 15, in <module>
    from habitat.utils.visualizations import utils
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/utils.py", line 20, in <module>
    cv2 = try_cv2_import()
  File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/utils.py", line 89, in try_cv2_import
    import cv2
ImportError: /usr/lib/x86_64-linux-gnu/libGL.so.1: undefined symbol: _glapi_tls_Current
Segmentation fault (core dumped)

yhshi9898, 2024-05-29:
> > Just in case someone needs this: here is a solution for those who don't want to use docker and happen to use a server that does not install `EGL*` along with NVIDIA driver by default. (cross-posted to #851 as well!)
> > 
> > 1. run `nvidia-smi` to check the version of your driver
> > 2. download the driver installation binary from NVIDIA directly. You can do this by searching `nvidia driver <version-id>` on google. The file should be `NVIDIA-Linux-x86_64-<version-id>.run`
> > 3. extract the binary without running. This will create a folder named `NVIDIA-Linux-x86_64-<version-id>`
> > 
> > ```shell
> > sh NVIDIA-Linux-x86_64-<version-id>.run --extract-only   
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 4. create some simlinks
> > 
> > ```shell
> > cd NVIDIA-Linux-x86_64-<version-id>
> > ln -s ./libEGL.so.<version-id>  libEGL.so.1 # note: there might be a libEGL_nvidia*, do not symlink it as libEGL.so.1 instead!
> > ln -s ./libGL.so.<version> libGL.so.1      
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 5. Add this folder to your LD search path
> > 
> > ```
> > export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > 6. Check your habitat linker (Thanks @erikwijmans for [this](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)!)
> > 
> > ```shell
> > cd
> > ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Note that `libEGL.so.1`, `libOpenGL.so.0` and `libGLdispatch.so.0` should be all pointed to the ones in `/path/to/NVIDIA-Linux-x86_64-<version-id>`.
> > 
> > 7. You're good to go! Just follow [this](/path/to/NVIDIA-Linux-x86_64-%3Cversion-id%3E) to verify your `habitat-sim` installation.
> 
> That seems like I still can not run well. Following is my wrong message. That seems like Ican not use opencv now. Traceback (most recent call last): File "/data5/xrh/navigation/habitat3/habitat-lab/examples/example.py", line 9, in import habitat.gym # noqa: F401 File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/**init**.py", line 9, in from habitat.core.benchmark import Benchmark File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/benchmark.py", line 20, in from habitat.core.env import Env File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/env.py", line 32, in from habitat.tasks.registration import make_task File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/registration.py", line 28, in _try_register_rearrange_task() File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/**init**.py", line 12, in _try_register_rearrange_task import habitat.tasks.rearrange.actions.actions File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/actions.py", line 17, in from habitat.tasks.rearrange.actions.articulated_agent_action import ( File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/articulated_agent_action.py", line 2, in from habitat.tasks.rearrange.rearrange_sim import RearrangeSim File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 35, in from habitat.datasets.rearrange.rearrange_dataset import RearrangeEpisode File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 18, in from habitat.datasets.pointnav.pointnav_dataset import PointNavDatasetV1 File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 16, in from habitat.tasks.nav.nav import ( File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/nav/nav.py", line 42, in from habitat.utils.visualizations import fog_of_war, maps File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/**init**.py", line 7, in from habitat.utils.visualizations import maps, utils File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/maps.py", line 15, in from habitat.utils.visualizations import utils File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/utils.py", line 20, in cv2 = try_cv2_import() File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/utils.py", line 89, in try_cv2_import import cv2 ImportError: /usr/lib/x86_64-linux-gnu/libGL.so.1: undefined symbol: _glapi_tls_Current Segmentation fault (core dumped)

Hello, I encountered the same issue as you. Did you manage to resolve it? Thank you!

2927803072, 2025-04-02:
> ive dataset  changed to default successfully.
> I1003 13:42:57.916199 29584 MetadataMediator.cpp:183] ::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to [...]

Hello, can you tell me how to solve this problem?

shuowang178, 2025-04-03:
> You shouldn't need to manually install them, they are installed as part of a normal driver install

EGL: Failed to get EGL display: Success
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
EGL: Failed to get EGL display: Success
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context
May I ask how to solve this problem, which has troubled me for a long time, thank you very much.

wlll123456, 2025-06-07:
> > > Just in case someone needs this: here is a solution for those who don't want to use docker and happen to use a server that does not install `EGL*` along with NVIDIA driver by default. (cross-posted to [#851](https://github.com/facebookresearch/habitat-sim/issues/851) as well!)以防万一有人需要这个：对于那些不想使用 docker 并且碰巧使用默认不安装 `EGL*` 和 NVIDIA 驱动程序的服务器的人来说，这里有一个解决方案。（也交叉发布到 [#851](https://github.com/facebookresearch/habitat-sim/issues/851) ！
> > > 
> > > 1. run `nvidia-smi` to check the version of your driver运行 `nvidia-smi` 检查驱动程序的版本
> > > 2. download the driver installation binary from NVIDIA directly. You can do this by searching `nvidia driver <version-id>` on google. The file should be `NVIDIA-Linux-x86_64-<version-id>.run`直接从 NVIDIA 下载驱动程序安装二进制文件。您可以通过在 google 上搜索 `nvidia 驱动程序 <version-id>` 来执行此作。该文件应为 `NVIDIA-Linux-x86_64-<version-id>.run`
> > > 3. extract the binary without running. This will create a folder named `NVIDIA-Linux-x86_64-<version-id>`在不运行的情况下提取二进制文件。这将创建一个名为 `NVIDIA-Linux-x86_64-<version-id>`
> > > 
> > > sh NVIDIA-Linux-x86_64-<version-id>.run --extract-only   
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 
> > > 4. create some simlinks  创建一些 simlink
> > > 
> > > cd NVIDIA-Linux-x86_64-<version-id>
> > > ln -s ./libEGL.so.<version-id>  libEGL.so.1 # note: there might be a libEGL_nvidia*, do not symlink it as libEGL.so.1 instead!
> > > ln -s ./libGL.so.<version> libGL.so.1      
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 
> > > 5. Add this folder to your LD search path将此文件夹添加到您的 LD 搜索路径
> > > 
> > > ```
> > > export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > 
> > > 6. Check your habitat linker (Thanks [@erikwijmans](https://github.com/erikwijmans) for [this](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)!)检查你的栖息地链接器（谢谢[这个 ](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)！
> > > 
> > > cd
> > > ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Note that `libEGL.so.1`, `libOpenGL.so.0` and `libGLdispatch.so.0` should be all pointed to the ones in `/path/to/NVIDIA-Linux-x86_64-<version-id>`.请注意，`libEGL.so.1`、`libOpenGL.so.0` 和 `libGLdispatch.so.0` 都应该指向 `/path/to/NVIDIA-Linux-x86_64-<version-id>` 中的那些。
> > > 
> > > 7. You're good to go! Just follow [this](/path/to/NVIDIA-Linux-x86_64-%3Cversion-id%3E) to verify your `habitat-sim` installation.你准备好了！只需按照[此](/path/to/NVIDIA-Linux-x86_64-<version-id>)内容验证您的 `habitat-sim` 安装即可。
> > 
> > 
> > That seems like I still can not run well. Following is my wrong message. That seems like Ican not use opencv now. Traceback (most recent call last): File "/data5/xrh/navigation/habitat3/habitat-lab/examples/example.py", line 9, in import habitat.gym # noqa: F401 File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/**init**.py", line 9, in from habitat.core.benchmark import Benchmark File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/benchmark.py", line 20, in from habitat.core.env import Env File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/env.py", line 32, in from habitat.tasks.registration import make_task File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/registration.py", line 28, in _try_register_rearrange_task() File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/**init**.py", line 12, in _try_register_rearrange_task import habitat.tasks.rearrange.actions.actions File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/actions.py", line 17, in from habitat.tasks.rearrange.actions.articulated_agent_action import ( File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/articulated_agent_action.py", line 2, in from habitat.tasks.rearrange.rearrange_sim import RearrangeSim File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 35, in from habitat.datasets.rearrange.rearrange_dataset import RearrangeEpisode File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 18, in from habitat.datasets.pointnav.pointnav_dataset import PointNavDatasetV1 File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 16, in from habitat.tasks.nav.nav import ( File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/nav/nav.py", line 42, in from habitat.utils.visualizations import fog_of_war, maps File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/**init**.py", line 7, in from habitat.utils.visualizations import maps, utils File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/maps.py", line 15, in from habitat.utils.visualizations import utils File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/utils.py", line 20, in cv2 = try_cv2_import() File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/utils. [... cut ...] ls/visualizations/utils.py“，第 20 行，在 cv2 = try_cv2_import（） 文件中 ”/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/utils.py“，第 89 行， 在 try_cv2_import import cv2 ImportError： /usr/lib/x86_64-linux-gnu/libGL.so.1： undefined symbol： _glapi_tls_Current Segmentation fault （core dumped）
> 
> Hello, I encountered the same issue as you. Did you manage to resolve it? Thank you!您好，我遇到了与您相同的问题。你设法解决了吗？谢谢！

https://github.com/facebookresearch/habitat-sim/issues/1511#issuecomment-2137381916

wlll123456, 2025-06-07:
> > > > Just in case someone needs this: here is a solution for those who don't want to use docker and happen to use a server that does not install  along with NVIDIA driver by default. (cross-posted to [#851](https://github.com/facebookresearch/habitat-sim/issues/851) as well!)`EGL*`
> > > > 
> > > > 1. run  to check the version of your driver`nvidia-smi`
> > > > 2. download the driver installation binary from NVIDIA directly. You can do this by searching  on google. The file should be `` nvidia driver <version-id>``NVIDIA-Linux-x86_64-<version-id>.run ``
> > > > 3. extract the binary without running. This will create a folder named `NVIDIA-Linux-x86_64-<version-id>`
> > > > 
> > > > sh NVIDIA-Linux-x86_64-.run --extract-only
> > > > 
> > > > 4. create some simlinks
> > > > 
> > > > cd NVIDIA-Linux-x86_64-
> > > > ln -s ./libEGL.so.  libEGL.so.1 # note: there might be a libEGL_nvidia*, do not symlink it as libEGL.so.1 instead!
> > > > ln -s ./libGL.so. libGL.so.1
> > > > 
> > > > 5. Add this folder to your LD search path
> > > > 
> > > > ```
> > > > export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > 
> > > > 6. Check your habitat linker (Thanks [@erikwijmans](https://github.com/erikwijmans) for [this](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)!)
> > > > 
> > > > cd
> > > > ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.**file**)")
> > > > Note that ,  and  should be all pointed to the ones in .`` libEGL.so.1``libOpenGL.so.0``libGLdispatch.so.0``/path/to/NVIDIA-Linux-x86_64-<version-id> ``
> > > > 
> > > > 7. You're good to go! Just follow [this](/path/to/NVIDIA-Linux-x86_64-%3Cversion-id%3E) to verify your  installation.`habitat-sim`
> > > 
> > > 
> > > That seems like I still can not run well. Following is my wrong message. That seems like Ican not use opencv now. Traceback (most recent call last): File "/data5/xrh/navigation/habitat3/habitat-lab/examples/example.py", line 9, in import habitat.gym # noqa: F401 File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/**init**.py", line 9, in from habitat.core.benchmark import Benchmark File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/benchmark.py", line 20, in from habitat.core.env import Env File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/env.py", line 32, in from habitat.tasks.registration import make_task File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/registration.py", line 28, in _try_register_rearrange_task() File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/**init**.py", line 12, in _try_register_rearrange_task import habitat.tasks.rearrange.actions.actions File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/actions.py", line 17, in from habitat.tasks.rearrange.actions.articulated_agent_action import ( File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/actions/articulated_agent_action.py", line 2, in from habitat.tasks.rearrange.rearrange_sim import RearrangeSim File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/rearrange/rearrange_sim.py", line 35, in from habitat.datasets.rearrange.rearrange_dataset import RearrangeEpisode File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/rearrange/rearrange_dataset.py", line 18, in from habitat.datasets.pointnav.pointnav_dataset import PointNavDatasetV1 File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/datasets/pointnav/pointnav_dataset.py", line 16, in from habitat.tasks.nav.nav import ( File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/tasks/nav/nav.py", line 42, in from habitat.utils.visualizations import fog_of_war, maps File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/**init**.py", line 7, in from habitat.utils.visualizations import maps, utils File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/maps.py", line 15, in from habitat.utils.visualizations import utils File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/utils/visualizations/utils.py", line 20, in cv2 = try_cv2_import() File "/data5/xrh/navigation/habitat3/habitat-lab/habitat-lab/habitat/core/utils.py", line 89, in try_cv2_import import cv2 ImportError: /usr/lib/x86_64-linux-gnu/libGL.so.1: undefined symbol: _glapi_tls_Current Segmentation fault (core dumped)
> > 
> > 
> > Hello, I encountered the same issue as you. Did you manage to resolve it? Thank you!
> > Hello, I encountered the same issue as you. Did you manage to resolve it? Thank you!

Hello, I encountered the same issue as you. Did you manage to resolve it? Thank you!

KindredSpirithub, 2026-02-05:
> Just in case someone needs this: here is a solution for those who don't want to use docker and happen to use a server that does not install `EGL*` along with NVIDIA driver by default. (cross-posted to [#851](https://github.com/facebookresearch/habitat-sim/issues/851) as well!)
> 
> 1. run `nvidia-smi` to check the version of your driver
> 2. download the driver installation binary from NVIDIA directly. You can do this by searching `nvidia driver <version-id>` on google. The file should be `NVIDIA-Linux-x86_64-<version-id>.run`
> 3. extract the binary without running. This will create a folder named `NVIDIA-Linux-x86_64-<version-id>`
> 
> sh NVIDIA-Linux-x86_64-<version-id>.run --extract-only   
> 4. create some simlinks
> 
> cd NVIDIA-Linux-x86_64-<version-id>
> ln -s ./libEGL.so.<version-id>  libEGL.so.1 # note: there might be a libEGL_nvidia*, do not symlink it as libEGL.so.1 instead!
> ln -s ./libGL.so.<version> libGL.so.1      
> 5. Add this folder to your LD search path
> 
> ```
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/path/to/NVIDIA-Linux-x86_64-<version-id>
> ```
> 
> 6. Check your habitat linker (Thanks [@erikwijmans](https://github.com/erikwijmans) for [this](https://github.com/facebookresearch/habitat-sim/issues/288#issuecomment-637756435)!)
> 
> cd
> ldd $(python -c "import habitat_sim; print(habitat_sim._ext.habitat_sim_bindings.__file__)")
> Note that `libEGL.so.1`, `libOpenGL.so.0` and `libGLdispatch.so.0` should be all pointed to the ones in `/path/to/NVIDIA-Linux-x86_64-<version-id>`.
> 
> 7. You're good to go! Just follow [this](/path/to/NVIDIA-Linux-x86_64-%3Cversion-id%3E) to verify your `habitat-sim` installation.

I have solved the problem "WindowlessEglApplication::tryCreateContext(): cannot get default EGL display" by this method，thanks a lot.  And I would like to suggest to add "ln -s libEGL_nvidia.so.<driver_version> libEGL_nvidia.so.0" , otherwise may met a new problem "Segmentation fault (core dumped) "