facebookresearch/habitat-sim #1512: How to generate camera poses?

zsh2000, 2021-10-04:
Thanks for the awesome tool! 

I have successfully rendered RGB images and semantic labels on Replica dataset using 

`python examples/example.py --save_png --semantic_sensor`

I want to get the corresponding camera poses (better in the format of extrinsic matrix and instrinsic matrix) together with the rendered RGB images and semantic labels. But there is no options in "example.py" enabling the generation of poses.

Could you please tell me how to do it?

closed 2022-01-26 by dhruvbatra

cbyan1003, 2022-10-07:
> Thanks for your reply! I'll try it out.

Could you please tell me how to get the camera pose in details?
And also I meet a problem, when I use `--semantic sensor` to get the output semantic image from replica dataset, but the ouput is wrong. How to generate semantic image? I guess my setting is wrong.

oskar0812, 2023-04-12:
Hi, I was wondering if you solved this problem? I also encountered the same problem