facebookresearch/habitat-sim #1513: Habita 2.0

dipanjan06, 2021-10-04:
Thanks for publishing a great paper https://arxiv.org/pdf/2106.14405.pdf. I am planning to use a mobile-based manipulator like the Fetch robot shown in your paper. However, I do not find any code base for the same. Have you published the the code base for Habitat 2.0 mentioned in your paper? It would be greart help if you guide me on how to use the simulator with the Fetch robot enabled. Thanks in advance.

dhruvbatra, 2021-10-04:
See footnote 3 in the paper. We have not yet released large parts of the code. We are working on it and a release will happen in the coming months.

closed 2021-10-04 by dhruvbatra

tangyuq, 2021-12-30:
@dhruvbatra Any plan for Habitat 2.0 release?  Thanks.