facebookresearch/habitat-sim #1562: Coordinate transformation in Habitat

yhymason, 2021-11-10:
Hi there, I have been trying to get transformations from coordinates in the camera frame to Habitat's world frame. However, as far as I understand, Habitat's world axes are in (-z, y, x) where -z is the forward direction of the agent. All the points in the camera frame are in the conventional (x, y, z) system where y is pointing out of the camera. Say I want to use the following info to transform the points in my camera frame, how should I carry out the transformations? 
```
sensor_states['depth'].rotation
sensor_states['depth'].position
and point cloud data created from depth observations from Habitat-sim
```
I have tried directly creating a 4 * 4 transformation matrix with the camera's rotation and position but things don't look right in the end. Am I missing something here? Here's how my transformation matrix looks:
```
rotation = R.from_quat(
      [
          camera_rotation[-1],
          camera_rotation[0],
          camera_rotation[1],
          camera_rotation[2],
      ]
  )
transformation_mat = np.zeros((4, 4))
transformation_mat[0:3, 0:3] = rotation.as_matrix()
transformation_mat[:, 3] = np.transpose(np.append(camera_position, [[1.0]]))
```

aclegg3, 2021-11-10:
Hey @yhymason,

You may want to take a look at [Camera.unproject](https://aihabitat.org/docs/habitat-sim/habitat_sim.gfx.Camera.html#unproject) from the API. If this doesn't directly help, it should allow you to trace the transformations in the code to construct your own solution.

yhymason, 2021-11-11:
@aclegg3 thanks, I took a look at [this](https://aihabitat.org/docs/habitat-lab/view-transform-warp.html) instead. I managed to follow it and reproduce world -> camera and camera -> world transformations as follows:
```
from scipy.spatial.transform import Rotation as R
import numpy as np
coeffs = np.array([-0.0019395975396037102, 0.48101964592933655, -0.002401920733973384, 0.8767043948173523])
position = np.array([-10.305132865905762, 1.3434221744537354, -4.843066692352295])
rotation = R.from_quat(
    [
        coeffs[0],
        coeffs[1],
        coeffs[2],
        coeffs[3],
    ]
)
# World to camera
transformation_mat = np.zeros((4, 4))
transformation_mat[0:3, 0:3] = rotation.as_matrix()
transformation_mat[:, 3] = np.transpose(np.append(position, [[1.0]]))
res1 = np.matmul(np.linalg.inv(transformation_mat), np.append(position, [[1.0]]))
transformation_mat = np.zeros((4, 4))
# Camera to world
transformation_mat[0:3, 0:3] = rotation.as_matrix()
transformation_mat[:, 3] = np.transpose(np.append(position, [[1.0]]))
res2 = np.matmul(transformation_mat, np.append([0, 0, 0], [[1.0]]))

res1 is [0, 0, 0, 1]
and res2 is equal to position
```
However, when I tried to reproduce the same transformations on my point cloud data (which is in the camera frame with z-axis pointing out of the camera) I could get correct transformation results. Here is what I did:
```
rotation = R.from_quat(
    [
        camera_quaternion[0],
        camera_quaternion[1],
        camera_quaternion[2],
        camera_quaternion[3],
    ]
)
transformation_mat = np.zeros((4, 4))
transformation_mat[0:3, 0:3] = rotation.as_matrix()
transformation_mat[:, 3] = np.transpose(np.append(camera_position, [[1.0]]))
print(transformation_mat)
points = pcd.points
new_points = []
for point in points:
    new_point = np.matmul(
        transformation_mat)
        np.append(
            point,
            [[1.0]],
        ),
    )
    new_points.append(new_point[0:3])
```
Would please take a look and tell me if anything is wrong?

@erikwijmans FYI

erikwijmans, 2021-11-11:
> All the points in the camera frame are in the conventional (x, y, z) system where y is pointing out of the camera.

Habitat uses the conventional graphics coordinate frame for the camera, where the camera looks along -z, +y is up, and +x is right. This is the same coordinate system as the agent -- which is why we use it for the agent :).

```
# World to camera
transformation_mat = np.zeros((4, 4))
transformation_mat[0:3, 0:3] = rotation.as_matrix()
transformation_mat[:, 3] = np.transpose(np.append(position, [[1.0]]))
```
The transformation_mat you are constructing here is not world to camera, it's camera to world (this is stated in the tutorial you linked). This is confirmed by you doing `np.matmul(np.linalg.inv(transformation_mat), np.append(position, [[1.0]]))` to get [0,0,0,1]. If transformation_mat was already world to camera, then you won't need the inversion.

```
However, when I tried to reproduce the same transformations on my point cloud data (which is in the camera frame with z-axis pointing out of the camera)
```
You'll likely need to apply a rotation to the point cloud before applying the world to camera transformation to correct for these differences in axes orientations.

yhymason, 2021-11-11:
@erikwijmans thanks for pointing `If transformation_mat was already world to camera, then you won't need the inversion.` out. However, I am not sure about `apply a rotation to the point cloud` since as for as I understand the cloud points are already in the camera's local frame. I don't have any additional rotational information about the cloud points. According to [this documentation](http://www.open3d.org/docs/0.7.0/python_api/open3d.geometry.create_point_cloud_from_depth_image.html), I am pretty confident the generated points are in the camera frame (where the camera looks along -z, +y is up, and +x is right). Can you elaborate on what kind of additional rotation I need to apply to the points?

erikwijmans, 2021-11-12:
Did you generate this point cloud from habitat or it from elsewhere? The line "which is in the camera frame with z-axis pointing out of the camera" made me think it's from elsewhere, but perhaps you mean the z-axis points into the camera? If it is from habitat then it should already be in the right coordinate frame.

yhymason, 2021-11-12:
@erikwijmans is there an existing method to generate point clouds from Habitat Sim or API? would you please show me? it'd be great if I can use something directly from Habitat. To your question, I am using open3d methods to generate the point clouds

erikwijmans, 2021-11-12:
There isn't one, but point cloud from RGB-D is very easy. The tutorial you linked generates a point cloud in it in fact. `xy_c0` is a point cloud in camera space. If you multiply that by `T_world_camera0` you'll get a point cloud in world space.

yhymason, 2021-11-13:
@erikwijmans I got everything to work after flipping the axes in the camera frame. Cheers.

closed 2021-11-13 by yhymason

arjung128, 2022-02-28:
```
T_world_camera0[0:3,0:3] = rotation_0
T_world_camera0[0:3,3] = translation_0
```
This is the form of the extrinsic parameter matrix from [here](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html), which maps world coordinates to camera coordinates. 
However, in the tutorial, this matrix is used to map camera coordinates to world coordinates. 
Is this not a contradiction? What am I missing?