facebookresearch/habitat-sim #1566: Issues with running lighting_tutorial

sparsh-umd, 2021-11-16:
I created a new conda environment(pythpn3.8.10 & cmake3.4.10) & installed habitat-sim with `conda install habitat-sim withbullet -c conda-forge -c aihabitat`. Then I ran 'python3 setup.py --bullet install'.

When I try to run the lighting_setup tutorial with `python3 examples/tutorials/lighting_tutorial.py`, I get the following error:
`ImportError: generic_type: type "SceneNode" referenced unknown base type "Magnum::SceneGraph::Object<Magnum::SceneGraph::BasicTranslationRotationScalingTransformation3D<float> >"
`

Thanks for the help!

erikwijmans, 2021-11-18:
```
installed habitat-sim with conda install habitat-sim withbullet -c conda-forge -c aihabitat. Then I ran 'python3 setup.py --bullet install'.
```

Don't install habitat-sim via conda and then via setup.py. That may end up causing a version mismatch between habitat and magnum (which seems like what happened here). Do only one or the other.

sparsh-umd, 2021-11-18:
Thanks for the reply @erikwijmans!

When I ran `lighting_setup.py` before installing habitat via `python3 setup.py --bullet install` I was getting something like `system is configured to run bullet but bullet isn't installed`. Hence, I ran the above command with setup.py, hoping that it'd solve the issue (Also, I thought I came across something like 'bullet physics engine doesn't get automatically installed with conda install').

Now, I tried to create a new conda environment & installed habitat-sim only with conda. I was able to run the testing with `habitat-viewer`, but lighting_tutorial.py isn't running. I'm getting a `Segmentation fault (core dumped)` error. The following is the complete terminal output:

WARNING: Logging before InitGoogleLogging() is written to STDERR
I1118 00:03:13.779678 12889 ManagedFileBasedContainer.h:210] <Dataset>::convertFilenameToPassedExt : Filename : default changed to proposed scene_dataset_config.json filename : default.scene_dataset_config.json
I1118 00:03:13.779711 12889 AttributesManagerBase.h:365] <Dataset>::createFromJsonOrDefaultInternal : Proposing JSON name : default.scene_dataset_config.json from original name : default | This file  does not exist.
I1118 00:03:13.779958 12889 AssetAttributesManager.cpp:120] Asset attributes (capsule3DSolid : capsule3DSolid_hemiRings_4_cylRings_1_segments_12_halfLen_0.75_useTexCoords_false_useTangents_false) created and registered.
I1118 00:03:13.780009 12889 AssetAttributesManager.cpp:120] Asset attributes (capsule3DWireframe : capsule3DWireframe_hemiRings_8_cylRings_1_segments_16_halfLen_1) created and registered.
I1118 00:03:13.780047 12889 AssetAttributesManager.cpp:120] Asset attributes (coneSolid : coneSolid_segments_12_halfLen_1.25_rings_1_useTexCoords_false_useTangents_false_capEnd_true) created and registered.
I1118 00:03:13.780071 12889 AssetAttributesManager.cpp:120] Asset attributes (coneWireframe : coneWireframe_segments_32_halfLen_1.25) created and registered.
I1118 00:03:13.780082 12889 AssetAttributesManager.cpp:120] Asset attributes (cubeSolid : cubeSolid) created and registered.
I1118 00:03:13.780093 12889 AssetAttributesManager.cpp:120] Asset attributes (cubeWireframe : cubeWireframe) created and registered.
I1118 00:03:13.780120 12889 AssetAttributesManager.cpp:120] Asset attributes (cylinderSolid : cylinderSolid_rings_1_segments_12_halfLen_1_useTexCoords_false_useTangents_false_capEnds_true) created and registered.
I1118 00:03:13.780143 12889 AssetAttributesManager.cpp:120] Asset attributes (cylinderWireframe : cylinderWireframe_rings_1_segments_32_halfLen_1) created and registered.
I1118 00:03:13.780156 12889 AssetAttributesManager.cpp:120] Asset attributes (icosphereSolid : icosphereSolid_subdivs_1) created and registered.
I1118 00:03:13.780166 12889 AssetAttributesManager.cpp:120] Asset attributes (icosphereWireframe : icosphereWireframe_subdivs_1) created and registered.
I1118 00:03:13.780184 12889 AssetAttributesManager.cpp:120] Asset attributes (uvSphereSolid : uvSphereSolid_rings_8_segments_16_useTexCoords_false_useTangents_false) created and registered.
I1118 00:03:13.780198 12889 AssetAttributesManager.cpp:120] Asset attributes (uvSphereWireframe : uvSphereWireframe_rings_16_segments_32) created and registered.
I1118 00:03:13.780205 12889 AssetAttributesManager.cpp:108] ::constructor : Built default primitive asset templates : 12
I1118 00:03:13.780763 12889 SceneDatasetAttributesManager.cpp:36] File (default) not found, so new default dataset attributes created and registered.
I1118 00:03:13.780778 12889 MetadataMediator.cpp:127] ::createSceneDataset : Dataset default successfully created.
I1118 00:03:13.780786 12889 AttributesManagerBase.h:365] <Physics Manager>::createFromJsonOrDefaultInternal : Proposing JSON name : ./data/default.physics_config.json from original name : ./data/default.physics_config.json | This file  exists.
I1118 00:03:13.781102 12889 PhysicsAttributesManager.cpp:26] JSON Configuration File (./data/default.physics_config.json) based physics manager attributes created and registered.
I1118 00:03:13.781116 12889 MetadataMediator.cpp:212] ::setActiveSceneDatasetName : Previous active dataset  changed to default successfully.
I1118 00:03:13.781119 12889 MetadataMediator.cpp:183] ::setCurrPhysicsAttributesHandle : Old physics manager attributes  changed to ./data/default.physics_config.json successfully.
I1118 00:03:13.781123 12889 MetadataMediator.cpp:68] ::setSimulatorConfiguration : Set new simulator config for scene/stage : data/scene_datasets/habitat-test-scenes/van-gogh-room.glb and dataset : default which is currently active dataset.
Renderer: AMD RENOIR (DRM 3.42.0, 5.14.8-051408-lowlatency, LLVM 12.0.0) by AMD
OpenGL version: 4.6 (Core Profile) Mesa 21.0.3
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    mesa-implementation-color-read-format-dsa-explicit-binding
    mesa-dsa-createquery-except-pipeline-stats
    mesa-forward-compatible-line-width-range
[... cut ...]
I1118 00:03:13.817132 12889 AttributesManagerBase.h:365] <Stage Template>::createFromJsonOrDefaultInternal : Proposing JSON name : data/scene_datasets/habitat-test-scenes/van-gogh-room.stage_config.json from original name : data/scene_datasets/habitat-test-scenes/van-gogh-room.glb | This file  does not exist.
E1118 00:03:13.817241 12889 StageAttributesManager.cpp:86] ::registerObjectFinalize : Render asset template handle : data/scene_datasets/habitat-test-scenes/van-gogh-room.glb specified in stage template with handle : data/scene_datasets/habitat-test-scenes/van-gogh-room.glb does not correspond to any existing file or primitive render asset.  Aborting. 
Segmentation fault (core dumped)

aclegg3, 2021-11-18:
`data/scene_datasets/habitat-test-scenes/van-gogh-room.glb does not correspond to any existing file or primitive render asset. Aborting.`
Looks like you need to download the example assets as described [here](https://github.com/facebookresearch/habitat-sim#testing).

sparsh-umd, 2021-11-18:
Thanks for the reply @aclegg3! It helps.

I actually ran that command & downloaded the example assets. It got downloaded as `data/versioned_data/habitat_test_scenes_1.0/` & the lighting_tutorial.py expects data as `data/scene_datasets/habitat-test-scenes/`. 
https://github.com/facebookresearch/habitat-sim/blob/3f385c4976dc2f8394f8e3f823ad21ef1d4cecbe/examples/tutorials/lighting_tutorial.py#L58

I changed the above line of code to `data/versioned_data/habitat_test_scenes_1.0/` & it works! You probably want to modify lighting_tutorial.py accordingly.

Thanks!

aclegg3, 2021-11-18:
Thanks @sparsh-umd, the datasets downloader should be creating `data/scene_datasets/habitat-test-scenes` as a symbolic link to `data/versioned_data/habitat_test_scenes_1.0/ `. Maybe this wasn't working properly on your system? I think your solution of using the absolute path makes sense, but beware other possible issues with tutorial code expecting that symlinks are created and valid.

closed 2021-11-18 by aclegg3

sparsh-umd, 2021-11-26:
Thanks for the reply @aclegg3 !

Oh, right! I understand what happened.

I downloaded the dataset on a different machine (The symlinks are working here) & just copied the data from that machine to the current machine I'm running the tutorial on.