facebookresearch/habitat-sim #1567: How to Add Blur to scene

sparsh-umd, 2021-11-16:
Is there a way to add motion blur to the scene?

Thanks!

Skylion007, 2021-11-17:
This can be easily implemented as a sensor noise model that uses Kornia, OpenCV or another python library that has an existing motion blur implementation.

sparsh-umd, 2021-11-17:
Thanks for the reply @Skylion007 !

1) I found some documentation about sensor noise models: https://aihabitat.org/docs/habitat-sim/habitat_sim.sensors.noise_models.html
But it doesn't has any info about how to use a custom noise model. Can you point me to documentation related to that, please?

2) I faced another problem which i described in issue 1566. Can you please look into that, too?

Thanks, 
Sparsh.

Skylion007, 2021-11-19:
To add a custom one, just create a new Python file here and follow the format of the other python files: ```https://github.com/facebookresearch/habitat-sim/tree/main/habitat_sim/sensors/noise_models``` you also need to modify the factory function in `__init__.py` and add entry for the added file.

closed 2022-01-26 by dhruvbatra