facebookresearch/habitat-sim #1570: Installing Habitat-Sim Package via Conda for Windows

jmeola42, 2021-11-17:
When I attempted to install Habitat-Sim through conda on my Windows machine, I received the following error:

>conda install habitat-sim withbullet -c conda-forge -c aihabitat
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - habitat-sim

Current channels:

  - https://conda.anaconda.org/conda-forge/win-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://conda.anaconda.org/aihabitat/win-64
  - https://conda.anaconda.org/aihabitat/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page."

When I went to the anaconda.org page, I found the conda installer for habitat-sim only supports Linux and OSX? Is there a way to install habitat-sim via conda for Windows or should I use pip instead (said it was experimental, so wanted to check first)

Skylion007, 2021-11-17:
We currently do not support Windows.

jmeola42, 2021-11-17:
> We currently do not support Windows. It may work on the Windows Subsystem for Linux2 with the newest release of Windows 10, but we have not tested it extensively.

At all or just with conda installation?

Skylion007, 2021-11-17:
At all.

jmeola42, 2021-11-17:
Should I install a VM with Linux and try to work on it there? Would that be more reliable?

Skylion007, 2021-11-19:
Assuming you can give the VM access to the GPU, then yes.

jmeola42, 2021-11-30:
I'm using VirtualBox and I have an NVidia GeForce GTX 1660 Ti. That should be sufficient for allowing the Virtual Machine access to my GPU, right?

jmeola42, 2021-11-30:
I guess I haven't found a way of giving the VM access to the GPU, since the AIHabitat Demo doesn't load the images on the Ubuntu VM

closed 2022-04-05 by dhruvbatra

anas-zafar, 2023-12-06:
hi @Skylion007 , do we have windows support now?

yaoling1997, 2024-10-30:
Sad it doesn't have windows support in 2024.